thotro/arduino-dw1000 #89: Why is "digitalWriteFast.h" used?

MikeS159, 2016-06-10:
As far as I can tell it is only used in a few places in DW1000.cpp, but I don't understand why as its no faster than `digitalWrite(pin, status)`.

Testing on an Arduino Uno with an oscilloscope code in the loop function setting a pin high then low with no delay gave a frequency of ~85kHz with digitalWrite and ~82kHz with digitalWriteFast.

Is there any reason it is used. If not it could be removed so the digitalWriteFast library isn't liked saving program space.

Rotzbua, 2016-06-10:
I think it was added a long time ago.

Did you used IDE 1.6?
I read somewhere at a forum that old with old IDE `digitalWrite` did intern many checks, so it was slow.
Since 1.6 it should be significant faster if *pin* is a `const` variable (information from that forum post, not officially read). I think that is maybe the reason for your results?

MikeS159, 2016-06-10:
That could explain it. I do vaguely remember something about improvements like that in 1.6.

closed 2016-06-10 by MikeS159