thotro/arduino-dw1000 #95: How to calculate the Time of Flight?

tn0432, 2016-06-27:
Hello!

I try to understand the way to calculate the Time of Flight in the code. In the file DW1000Ranging.cpp,
the calculation is made by:
DW1000Time round1 = (myDistantDevice->timePollAckReceived-myDistantDevice->timePollSent).wrap();
    DW1000Time reply1 = (myDistantDevice->timePollAckSent-myDistantDevice->timePollReceived).wrap();
    DW1000Time round2 = (myDistantDevice->timeRangeReceived-myDistantDevice->timePollAckSent).wrap();
    DW1000Time reply2 = (myDistantDevice->timeRangeSent-myDistantDevice->timePollAckReceived).wrap();
    
myTOF->setTimestamp((round1 * round2 - reply1 * reply2) / (round1 + round2 + reply1 + reply2));

What is the name of the formula? Hope anyone explain this point.

Thanks

Rotzbua, 2016-06-28:
I think this was the formula from the dw1000 data sheet. It is a modification of the simples ToF formula.

closed 2017-04-26 by Rotzbua

lrabe, 2018-08-28:
The formula is derived in Annex D of IEEE 802.15.8-2017, pages 307-312.

See here: https://www.decawave.com/decaforum/showthread.php?tid=1118&pid=3202#pid3202