thotro/arduino-dw1000 #99: list of part for dwm1000 arduino standalone

cyryllo, 2016-07-14:
Hello

Where can I find a list of needed parts and a description of board  dwm1000 arduino standalone?

MikeS159, 2016-07-14:
https://github.com/thotro/arduino-dw1000/wiki/Testbed-and-Adapter-board
There are schematics for getting a board for the DWM100 made in the [repository ](https://github.com/thotro/arduino-dw1000/tree/master/adapterBoard)too.

cyryllo, 2016-07-14:
I search info (list od parts) about this board https://github.com/thotro/arduino-dw1000/tree/master/adapterBoard/dwm1000%20arduino%20standalone

MikeS159, 2016-07-18:
That looks like a Pro Mini chip with the minimum components to make it work, and the DWM1000. Maybe @leosayous21 can confirm?

cyryllo, 2016-07-18:
I found in schemat file on Eagle ;)

closed 2017-04-26 by Rotzbua