thotro/arduino-dw1000 #106: How to group up Tag & Anchors using Network ID to work independently?

mic980000, 2016-08-07:
Dear All, I've managed to use the Ranging Tag&Anchor example sketch to print out some data. My question is, would that be possible to set different network ID to group up Tag&Anchors to work independently?

 Let say 1Tag 3Anchors as a group (Network 1) then i would have another set of group (Network 2) working in the same environment at the same time independently without interference each-other.  I've tried to change DW1000.setNetworkId(2); in void setup() but with no luck, anyone could shine a light on me.. THX :)

mic980000, 2016-08-10:
Hi Guys, I have successfully modified  "The Basic sender & receiver" to have a Group ID, basically if either side change the group ID the Anchor will stop receiving data.

mic980000, 2016-08-10:
But when I try to do the same in Ranging Tag & Anchor it seems going nowhere :(
I have made a start, see if anyone could make it work. Thanks

[RangingTag&Anchor.zip](https://github.com/thotro/arduino-dw1000/files/411194/RangingTag.Anchor.zip)

mic980000, 2016-08-11:
How to change this to make it work?? Anyone please help........

void DW1000RangingClass::startAsTag(char address[], const byte mode[]) {
	//save the address
	DW1000.convertToByte(address, _currentAddress);
	//write the address on the DW1000 chip
	DW1000.setEUI(address);
	Serial.print("device address: ");
	Serial.println(address);
	//we need to define a random short address:
	randomSeed(analogRead(0)); //< --------- Change This??
	_currentShortAddress[0] = random(0, 256); //< --------- Change This??
	_currentShortAddress[1] = random(0, 256); //< --------- Change This??

closed 2016-08-11 by mic980000

mertdluffy, 2022-12-05:
You can use channels, for example

>  setChannel(CHANNEL_5); 
for network 1 and
>  setChannel(CHANNEL_4); 
for network 2