thotro/arduino-dw1000 #107: Unstable operation with a new library

R2D2RU, 2016-08-10:
Hi.
I repeated device Arduino nano and Mega. 
[image]
[image]
everything worked. 
But after upgrading the library, devices stopped working.
Even a basic test  reads unstable data
What can be ???

R2D2RU, 2016-08-10:
Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: 00, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: E0FF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 1, version: 3, revision: 0

R2D2RU, 2016-08-10:
in the old library did change:

void DW1000RangingClass::initCommunication(unsigned int myRST, unsigned int mySS) {
	// reset line to the chip
	_RST              = myRST;
	_SS               = mySS;
	_resetPeriod      = DEFAULT_RESET_PERIOD;
	// reply times (same on both sides for symm. ranging)
	_replyDelayTimeUS = DEFAULT_REPLY_DELAY_TIME;
	//we set our timer delay
	_timerDelay       = DEFAULT_TIMER_DELAY;
	
	
//	DW1000.begin(1, myRST);
	DW1000.begin(0, myRST);
	DW1000.select(mySS);
}

Rotzbua, 2016-08-10:
> But after upgrading the library,
Version from to?

R2D2RU, 2016-08-10:
05.05.2016  11:42             5 916 digitalWriteFast.h
05.05.2016  11:42            52 724 DW1000.cpp
05.05.2016  11:42            25 061 DW1000.h
05.05.2016  11:42             3 919 DW1000Device.cpp
05.05.2016  11:42             2 663 DW1000Device.h
05.05.2016  11:42             4 641 DW1000Mac.cpp
05.05.2016  11:42             2 565 DW1000Mac.h
05.05.2016  11:42            29 739 DW1000Ranging.cpp
05.05.2016  11:42             5 597 DW1000Ranging.h
05.05.2016  11:42             4 367 DW1000Time.cpp
05.05.2016  11:42             2 974 DW1000Time.h

Rotzbua, 2016-08-10:
This is not really useful information. I suggested more a `last commit hash`.

My suggest is, that the irq pin is not right. There was a bug, that instead of the real pin the irq register was set. May look at your config.

R2D2RU, 2016-08-11:
IRQ PIN is connected to D2

Old library:

DW1000randing.cpp:
void DW1000RangingClass::initCommunication(unsigned int myRST, unsigned int mySS) {
	// reset line to the chip
	_RST              = myRST;
	_SS               = mySS;
	_resetPeriod      = DEFAULT_RESET_PERIOD;
	// reply times (same on both sides for symm. ranging)
	_replyDelayTimeUS = DEFAULT_REPLY_DELAY_TIME;
	//we set our timer delay
	_timerDelay       = DEFAULT_TIMER_DELAY;
	
	DW1000.begin(0, myRST);
	DW1000.select(mySS);
} 

DW1000.cpp:
void DW1000Class::begin(int irq, int rst) {
	// generous initial init/wake-up-idle delay
	delay(5);
	// start SPI
	SPI.begin();
	SPI.usingInterrupt(irq);
	// pin and basic member setup
	_rst        = rst;
	_irq        = irq;
	_deviceMode = IDLE_MODE;
	// attach interrupt
	attachInterrupt(_irq, DW1000Class::handleInterrupt, RISING);
} 
Interrupt handler is not beautiful, but it is correct.....

I think that the problem in the SPI speed.....

Rotzbua, 2016-08-11:
`DW1000.begin(0, myRST);`
0!=D2
=>`DW1000.begin(D2, myRST);`

R2D2RU, 2016-08-11:
No! This version of the library works!

R2D2RU, 2016-08-11:
oid DW1000Class::begin(int irq, int rst) {
	// generous initial init/wake-up-idle delay
	delay(5);
	// start SPI
	SPI.begin();
	SPI.usingInterrupt(irq);
	// pin and basic member setup
	_rst        = rst;
	_irq        = irq;
	_deviceMode = IDLE_MODE;
	// attach interrupt
	attachInterrupt(_irq, DW1000Class::handleInterrupt, RISING);
} 
pin D2 - interrupt 0

Rotzbua, 2016-08-11:
....
told you:

> My suggest is, that the irq pin is not right. There was a bug, that instead of the real pin the irq register was set. May look at your config.

So change from old => new version.

> `DW1000.begin(0, myRST);`
> 0!=D2
> =>`DW1000.begin(D2, myRST);`

R2D2RU, 2016-08-11:
What you write is true for the new library

R2D2RU, 2016-08-11:
In the old version  need to specify the interrupt number.

R2D2RU, 2016-08-11:
Has reduced the speed of SPI. Began working  the basic test. Randing not work.

Rotzbua, 2017-04-24:
@R2D2RU Is this still an issue?