thotro/arduino-dw1000 #108: Any Updtae in TDOA

maxicastle, 2016-08-10:
Hello Everyone,

Some of you can get some advance in one of these critical topics for this Library:

*Wired Clock Synchronization
*Wireless Clock synchronization
*TDoA System
*Multiple TAG Support without a cost in Update rate

Rotzbua, 2016-09-10:
I think this library is more a driver than an application. So the topics you mention should be an extra repository. Maybe you find some code if you search some papers from universities.

Slamoth, 2017-02-17:
DW1000 has a sync pin but DWM1000 does not have a sync pin. It says in DWM1000 documentation that wireless sync is possible but I could not find any information about how it can be done. The only clue I got is "reset the system clock counter." but I do not know how to do it either. Getting drowned in the documantation.

christoshadjiaslanis, 2017-02-18:
I am looking forward to doing wired synchronization. I looked over the Decawave paper on the wireless synchronization and it seems much less straight forward. So any TDOA stuff requiring synchronization at this point should probably be done on the DW1000. I also have this feeling that their wireless synchronization protocols are proprietary and they are not widely accessible. Also I think this is why the DWM1000 does not have the SYNC pin. Just a hunch.

Also I found an older version of the DWM1000 (https://openrtls.com/shop/product/dwm1000-ortls-moq-50-pcs-13?category=5) which does have the SYNC pin. I'm not sure if its compatible with this library though. If it is, it may make it easier to get straight to testing without having to print custom PCB adapter boards.

Slamoth, 2017-02-20:
I wrote to decawave. Will wait and see about the answer. I will share the answer when it comes.

maxicastle, 2017-02-20:
The problem is this, You need the Sync Pin (GPIO7) but also you need to provide the same 38.4MHz Clk signal to all anchors (in order to avoid clock violations), the EXTCLK (XTAL1) is not available in the DWM1000, because is connected directly to an 38.4 XTAL oscilator.

StrHero, 2017-03-03:
@Slamoth Looking forward to your reply.

Hope to be able to exchange learning with everyone.
My Facebook:Yeoman Ku

Slamoth, 2017-03-03:
Reply from decawave to the question of wireless sync option....
The DWM1000 could be used for TdoA  applications, but we recommend only for TAG purpose/applications. In a wired synchronisation scheme the DWM1000 is not suitable as Anchor as the externally sync pin available on DW1000  (pin 29) is not available on the DWM1000 (pin 4).  This is also stated in the DWM1000 datasheet. Information on wired synchronisation is available in our application note: APS007 Wired Sync RTLS with the DW1000,  which is attached (it is also available from our website.) Information on wireless synchronisation of the clocks is available, **but there is a fee attached as it contains IP.**

StrHero, 2017-03-04:
@Slamoth Thank you very much!
Is this mean that if we don't know the IP,then we'll not do the project based on wireless synchronisation?
Do you know the price of the IP of wireless synchronisation.

maxicastle, 2017-03-04:
I asked to decawave some months ago, The IP cost is more or less 35K USD really expensive, and only includes Documentation and pseudocodes. But something interesting the decawave expert tells me that even with the possibility to make a wireless sync, a wired sync system is alway more robust.

Slamoth, 2017-03-05:
Wireless sync is all about blink frames but you need LOS condition for that
to work as expected. Maxi is right about wired sync.

Slamoth, 2017-03-07:
This thing contains all TDOA in it. http://www.semiconductorstore.com/cart/pc/viewPrd.asp?idproduct=50835&utm_source=octopart&utm_medium=compref&utm_campaign=OCTPTEC&utm_term=SW1201_TDOA_RTLS_SOFTWARE

Slamoth, 2017-03-17:
Since this is not about open source I am closing the issue.

Rotzbua, 2017-04-26:
Please move the information about TDoA to the wiki. Thanks.
duplicates #47 #77 #108 #128 #130 #152

AlexisTM, 2018-03-27:
Reverse TDOA implementation by the Loco positioning system by Bitcraze for CrazyFlies, using the DWM1000.

Anchor software:

anchor.c: https://github.com/bitcraze/lps-node-firmware/blob/master/src/uwb_tdoa_anchor2.c
Tag software:

tag.c: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/src/lpsTdoaTag.c
tag.h: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/interface/lpsTdoaTag.h
config: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/src/locodeck.c

anthony211212, 2023-08-31:
Does anyone have any source code related to TDOA? Has anyone found any new developments regarding this topic?

AlexisTM, 2023-09-01:
@anthony211212 I had 16 robots running with this implementation
https://github.com/AlexisTM/MultilaterationTDOA

The board was a Bitcraze Roadrunner (using DW1000) sending the data out in this [custom firmware](https://github.com/umdlife/crazyflie-firmware/tree/u2_feature_custom_kalman) using this data format [Mavlink format](https://github.com/umdlife/roadrunner_mavlink)

You can "just" replicate [the Bitcraze algorithm](https://www.bitcraze.io/documentation/repository/lps-node-firmware/master/protocols/tdoa3_protocol/) and have a cheap solution.

anthony211212, 2023-09-01:
I would like to use ESP32 microcontroller and dw1000. Do you think it would be appropriate for this situation? Thanks.

AlexisTM, 2023-09-01:
Yes, you can, I had success from 8 anchors and we were using 16 for coverage of the area.

anthony211212, 2023-09-01:
Thank you for your message. Thotro library to calculate distance between anchor and tag with twr method and then you have performed distance measurement with tdoa method with a code similar to the one in the link you shared. Am I understanding it correctly? Because I didn't see a separate code file for anchor and tag.

anthony211212, 2023-09-01:
By the way I have  ESP32 DW1000 UWB (Ultra Wideband) Module and I ask for it.

AlexisTM, 2023-09-01:
Hi, that is incorrect. 
TDoA is a completely different concept. You could understand it as the **Anchors** does TWR between themselves;
The tags (whatever the amount) are **purely passive** and **listening** to the anchors communication.
From the anchor data received, the tags infers their position.

anthony211212, 2023-09-01:
I will check it right away. Thank you very much for your answers.

anthony211212, 2023-09-01:
Clearly, I'm new to this stuff, so do you have a chance to explain a bit more explicitly? For example, which codes do I need to upload separately for the tags and anchors? Thank you for your help and answers.

anthony211212, 2023-12-12:
@AlexisTM  Hey Alexis. I would like to ask which code files did you use for anchors and tags?

AlexisTM, 2023-12-13:
I used the Crazyflie Roadrunner, with a custom patch ([umdlife#crazyflie-firmware@u2_feature_custom_kalman](
https://github.com/umdlife/crazyflie-firmware/tree/u2_feature_custom_kalman)) to send out the TDoA measurement outside of the device [AlexisTM/roadrunner_mavlink](https://github.com/AlexisTM/roadrunner_mavlink) then compute the position using [AlexisTM/MultilaterationTDOA](https://github.com/AlexisTM/MultilaterationTDOA).

This was a while ago. The crazyflie patch would have to be redone.