thotro/arduino-dw1000 #109: Basic Connectivity test

Mani1520, 2016-08-12:
Hi All,

I am using Decawave DWM1000 with arduino uno to test the basic connectivity test program.
In my serial monitor i am geting below results

DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 28, version: 0, revision: 11
Unique ID: 90:CC:FF:83:EE:00:B0:E8
Network ID & Device Address: PAN: D3FF, Short Address: BED1
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 241, version: 15, revision: 10
Unique ID: 84:A0:C0:A4:9C:80:80:80
Network ID & Device Address: PAN: A880, Short Address: FFA3
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 217, version: 14, revision: 14
Unique ID: B9:BF:D1:5F:C4:C2:C8:C8
Network ID & Device Address: PAN: D796, Short Address: EC1B
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 202, version: 15, revision: 12
Unique ID: BD:C1:F0:60:7A:72:F2:FE
Network ID & Device Address: PAN: EAC7, Short Address: FF91
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 198, version: 14, revision: 8
Unique ID: B1:D4:D0:F6:DD:BB:F8:FF
Network ID & Device Address: PAN: 6FD5, Short Address: D8F5
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

Now I am working with two arduino uno for Basic sender and receiver code. But thats not working.

To convert the level I am using level sifter  TXB0106PWR.

Rotzbua, 2016-08-15:
Basic Connectivity test do not return success or fail.

device id should be constant, like Device ID: DECA - model: 1, version: 3, revision: 0

az-z, 2016-08-23:
@Rotzbua 
i'm getting 
[code]
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
[/code]
on UNO with the shifter. Do i read the above as a short on the wires somewhere?

closed 2017-04-24 by Rotzbua