thotro/arduino-dw1000 #112: ANCHOR to ANCHOR ranging

manomitbal, 2016-08-17:
Would it be possible to have dynamic ANCHOR-TAGS to have ANCHORs ranging with each other and sending those ranges to a TAG?

For example, if there are 3 ANCHORS A1, A2, A3 and 1 TAG, T1, we could read from T1:
Range from A1 to T1
Range from A2 to T1
Range from A3 to T1
Range from A1 to A2
Range from A1 to A3
Range from A2 to A3

This would greatly assist in trilateration/triangulation problems and help resolve the issue for multiple tags, since we would not require any.

In the `DW1000RangingClass::loop()` function, would it be possible to make the structure generic so as to have polling with ANCHORs as well?

Thanks

az-z, 2016-12-18:
>This would greatly assist in trilateration/triangulation problems and help resolve the issue for multiple tags, since we would not require any.
are you talking about peer-to-peer measurements?

>In the DW1000RangingClass::loop() function, would it be possible to make the structure generic so as to have polling with ANCHORs as well?
have you tried ? what are you findings?

berkerbatur, 2017-01-29:
@manomitbal you could unify anchor and tag sketches, and implement your own logic where each component polls to all others together with responding to other polls. But surely this will get your implementation complicated.