thotro/arduino-dw1000 #114: RESET DWM1000.....

R2D2RU, 2016-08-26:
I think part of the code is not correct

void DW1000Class::reset() {
.....................
		pinMode(_rst, INPUT);
......................
}

The correct code:
..................
	pinMode(_rst, INPUT_PULLUP);
.....................

Rotzbua, 2016-08-27:
No, datasheet says that not pullup rest.

R2D2RU, 2016-08-31:
You're wrong....

Aaron3142, 2016-08-31:
Rotzbua is right...
datasheet: "RSTn must never be pulled high by an external source"

dcellucci, 2016-11-30:
Hello, 

I ran the BasicConnectivityTest with RST first as INPUT then as INPUT_PULLUP, and I was able to get consistent state data from the RST line only when INPUT_PULLUP. Not sure what is happening, but the V2.10 User Guide for the DW1000 does not list any state requirements. (http://www.decawave.com/support/download/file/nojs/796)

I am using the Decawave Pro Mini Adapter with a 5V, 16 MHZ Pro Mini.

Rotzbua, 2016-11-30:
Plz read the `datasheet`  not the `user manual`!

dcellucci, 2016-11-30:
Thank you for the clarification. This still does not solve the fundamental problem: that this code, when used with the Decawave Pro Mini Adapter, does not produce an open-drain on the reset pin.

dcellucci, 2016-12-01:
I posted the question on Stack Overflow, and got this response: http://electronics.stackexchange.com/questions/272448/open-drain-through-schmitt-trigger/272477#272477

In response to that advice, I directly connected pin 9 to the RSTn pin of the Decawave, and everything is working. I'll let the board designer know that there is a problem with using a Schmitt trigger for that connection.

Rotzbua, 2016-12-01:
Ok, I did not recognized that you use a level shifter.. But I would not connect the DW RST pin to the 5V of the arduino. In your case I would recommend to use the software based triggered rest.

chipaudette, 2017-02-18:
I tried the solution posted above by @dcellucci .  I tried it on two boards.  It didn't work on either.  So, either my methods for the modification (connecting pin 9 to RST and cutting the trace from the level shifter to pin 9) were incorrect, or I damaged the parts, or @dcellucci got lucky.

It's frustrating when stuff doesn't work the way that one would like.

Chip

Rotzbua, 2017-04-24:
Problem is that the level shifter has states (0or1) and not floating. https://github.com/thotro/arduino-dw1000/issues/158#issuecomment-284151154

closed 2017-04-24 by Rotzbua