thotro/arduino-dw1000 #116: 3rd anchor refuses to associate

Phenioxrj, 2016-08-30:
Hey all!,

I'm having an issue when I try 3 anchors online to talk to 1 tag. Any two of the tags will work togther but when the third tag comes online, the entire program freezes. I'm running an arduino pro-mini (3.3 8MHZ) on a custom pcb. Im not sure if this is a hardware issue, or a software issue. Is there something I can change in the library or if i move up to an avr 2560 could this issue go away?

Thanks 
-Jules

from: 771C	 Range: 3.23 m	 RX power: -78.26 dBm
from: 6EF1	 Range: 3.60 m	 RX power: -77.34 dBm
from: 771C	 Range: 3.16 m	 RX power: -78.21 dBm
from: 6EF1	 Range: 3.65 m	 RX power: -77.32 dBm
from: 771C	 Range: 3.51 m	 RX power: -78.22 dBm
from: 6EF1	 Range: 3.61 m	 RX power: -77.32 dBm
from: 771C	 Range: 3.26 m	 RX power: -79.40 dBm
ranging init; 1 device added ! ->  short:6F6F
delete inactive device: 6F6F
ranging init; 1 device added ! ->  short:6F6F
delete inactive device: 6F6F
ranging init; 1 device added ! ->  short:6F6F
delete inactive device: 6F6F
ranging init; 1 device added ! ->  short:6F6F
delete inactive device: 6F6F
ranging init; 1 device added ! ->  short:6F6F
delete inactive device: 6F6F
ranging init; 1 device added ! ->  short:6F6F
delete inactive device: 6F6F

Nyuman, 2016-09-02:
I have the same problem like you. Have you found a solution for this problem since your post?

jack880019, 2016-09-08:
Hello @Phenioxrj , I'm also working on dwm1000. Everything works fine when I using the example in the library. Now I'am trying to have 1 tag and 3 Anchors and all anchors are going to connect to the computer but it seems that the example can just make the ranging process between one anchor and one tag don't know how to have tag to communicate to all the anchors automatically. Any Ideas?

soonweng111, 2016-11-19:
Hello @jack880019 I am also working on that too and having the same issues. Are there any solutions for that problem? Been trying to figure out for a few weeks but there is still no progress yet.

madams1337, 2016-11-23:
Hey guys,

@Phenioxrj @Nyuman 
If you get these values, there's probably no hardware fault since u got ranging results and also rssi. My guess is software.

We had the very same problem earlier this month. Our problem was caused by the Arduino IDE.
We figured out that versions <= 1.6.9 are working with this lib, higher versions caused the  problem you're describing. Maybe you can check that and give a lil' feedback.

flaviudan, 2016-11-27:
Hello,

For the last 1-2 weeks I am testing this project, and I have the same issue. The first two anchors seem to work fine but when I try to add the third I receive that error. 

Currently I am using Pro mini 3.3v version and external power supply for the DWM1000 chip.

I have tried to tinker with the library and also the above solution of downgrading the IDE(I am now running the 1.6.8 version), but nothing worked.

Am I missing something?

Please help

flaviudan, 2016-11-28:
Hello,

Good news, apparently I found a solution that works for me. It needs much more testing but for now works.

In the DW1000Ranging.cpp I have changed in the measurements part all the "<" with "<=", in the for loops.

In the tag part everything looks ok, but in the anchor part I receive 1 measurement, after that 4 "Not found" so there still is some room for improvement, or correct me if it is normal behavior.

Tomorrow I will try to connect the  fourth Anchor , so wish me luck.

Best regards

Slamoth, 2016-12-15:
Hello,
Have you found a solution for the third anchor yet ?
Which example are you using?

maxicastle, 2016-12-16:
Try to use Arduino 1.6.9 also this old version of the library

it works for me (Arduino Pro Mini 5V 16MHz  running at 3.3V)

maxicastle, 2016-12-16:
This One

Slamoth, 2016-12-16:
So you are using RangingTag and RangingAnchor code not the DW1000Ranging_TAG and DW1000Ranging_ANCHOR codes. I will give it a shot. This library is really good however there are some major drawbacks like allowing only 1 tag.

zohaib-arshad, 2017-01-12:
Hello @madams1337 , @maxicastle , @flaviudan .

We have a working tag with a single anchor. But adding multiple anchors induces the same connect/disconent problem faed by others.

Is it a power issue of a sofware problem?
How to resolve it?

Any help will be much appreciated.

cunglsang, 2017-01-12:
The current work around fix for me is aslo using arduino IDE <= 1.6.9 as @madams1337 said. Please make sure to delete all the dependancies when uninstalling the higher version of IDE (especially for windows users). Or just try it with the protable IDE version.

AliSiddiqui4, 2017-01-12:
^sorry to ask this nooby question but how do u delete dependencies? After installing 1.6.8 I am getting this error: 

avrdude: verification error, first mismatch at byte 0x0002
         0x8b != 0x8e
avrdude: verification error; content mismatch

Slamoth, 2017-02-01:
I think changing IDE version is not a real solution. We had problems before putting a 1000uf caps just after arduino pro mini 3.3V 8Mhz power output going to the DWM1000 chip. Power consumption is a real pain when sending uwb signals. My advice would be, first check how much power u need. Then optimize arduino for low power (cut off led) then turn of unneeded stuff like ADC on arduino to lower power as much as you can. The trick is not feeding the arduino, its feeding the dwm and other components.

khawajamechatronics, 2017-03-09:
Hi, I had same issue, any solution?

rbrtwtrs, 2017-03-14:
I also have the same issue. Hold any 1 of the three anchors in reset and the other two work. Hope someone has a fix.

commit 90133118 (2017-07-10): Merge pull request #214 from enacuavlab/multi_anchors_fix

fix third anchor loosing connection
seems fix #116 #178
closed 2017-07-10 by Rotzbua