thotro/arduino-dw1000 #117: DW1000 library and M0 board (ATMEL SAMD21G18)

sloubiou, 2016-08-30:
Hello,

I would like to use the library DW1000 on M0 board.
With some little change on example:
- BasicConnectivityTest run
- BasicSender doesn't run: DW1000Class::handleInterrupt() is never called and checking SPI library I don't understand how it could run???

On the other side I don"t understand on schematics why IRQ from the DW1000 is connected to Arduino IRQ input???

Thanks a lot for your suppport

Phenioxrj, 2016-08-30:
Could the IRQ line be hooked up to any digital line on the m0 since they all can be interrupts?

sloubiou, 2016-08-30:
I change interrupt pin from 2 to 3 because on 2 it was not running.
I test caalback with this SW and LED blink when I do a pin change state:

const byte ledPin = 13;
const byte interruptPin = 3;
volatile byte state = LOW;

void setup() {
  pinMode(ledPin, OUTPUT);
  pinMode(interruptPin, INPUT_PULLUP);
//  attachInterrupt(digitalPinToInterrupt(interruptPin), blink, CHANGE);
  attachInterrupt(interruptPin, blink, CHANGE);
}

void loop() {
  digitalWrite(ledPin, state);
}

void blink() {
  state = !state;
}

sloubiou, 2016-09-01:
I made some tests to sole the issue (I pool SYS_STATUS to track the emission) and in fact:
- first send I receive no interrup from the DW1000
- other I receive interrupt

Why?

khalil19, 2016-09-05:
Hi @sloubiou ,
I had the same problem with Arduino Due and now same problem with my arduino Zero.
Can you share your code (your test to track emission)?

sloubiou, 2016-09-06:
Finally I found what is the issue.
Issue is coming from SPI configuration...

Because clock is 48MHz on M0, 16MHz SPI fast and 2MHz SPI slow doesn't run.
You should configure (DW1000.cpp,) fast SPI (_fastSPI) to 12MHz ans slow SPI (_slowSPI) to 1.5MHz.

Now sender and receiver have discussion with interrupts.

Good luck

Phenioxrj, 2016-09-06:
@sloubiou Did you have to change any code in addition to the spi to get it to work on the zero?

sloubiou, 2016-09-07:
I change a lot of thing to understand the issue but finally only SPI is enought for me.
I'm working on M0 / M0 pro not on Zero but should be the same id micro frequency is 48MHz.

closed 2016-09-12 by sloubiou

HappyDr0id, 2017-01-17:
@sloubiou Hello, thanks a lot for your advices about the Arduino M0 Pro. I have this same board and I followed the testboard pins of the wiki : reset on the pin 9, MOSI on the pin 11, etc. But I do not manage to validate the BasicConnectivityTest sketch.

I only get :
Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

I changed the IRQ Pin to 3 and also changed the SPI max and min speed but nothing is working.
Could you please share your wiring diagram to be sure that I am not doing anything wrong ?

Thanks a lot in advance ;)

sloubiou, 2017-01-24:
Hello,

 

Please add a look on my web site :

http://www.aiill.fr/Geolocalisation_et_Integration_Arduino_DW1000.html

 

 

Sincères salutations

Simon Loubiou 
AiiLL Loubiou EIRL 
Tel : 07-82-76-05-47 
Tel : +33 782-760-547 
simon.loubiou@aiill.fr <mailto:simon.loubiou@aiill.fr>  
http://www.aiill.fr <http://www.aiill.fr/>

az-z, 2017-01-24:
> auth/AUWYKDJ-Wk_zpe2uC6WPiPWCSocClXoAks5rTPl-gaJpZM4Jwh03> . <

Simon
 Looks interesting. Do you have a  English version?

sloubiou, 2017-01-25:
No sorry

 

Sincères salutations

Simon Loubiou 
AiiLL Loubiou EIRL 
Tel : 07-82-76-05-47 
Tel : +33 782-760-547 
simon.loubiou@aiill.fr <mailto:simon.loubiou@aiill.fr>  
http://www.aiill.fr <http://www.aiill.fr/>