thotro/arduino-dw1000 #118: difference btw DW1000Ranging_TAG.ino and RangingTag.ino (resp DW1000Ranging_ANCHOR.ino and RangingAnchor.ino

wazaahhh, 2016-09-08:
I tried and used successfully both pairs of sketches to have 2 modules communicate together. However, RangingTag/Anchor.ino is way more precise than DW1000Ranging_TAG/ANCHOR. Yet the latter is made of much more streamlined code, and devices are uniquely identified, which is great.

Is there a reason for using one or another sketch pairs?

In both cases, anchors/tags are not super happy when I had a third device, either as anchor or tag. I gather from other issues that the code has not been written yet to accomplish  this task. I wonder if there is a to do list, which could help direct effort in this direction. Thanks.

Rotzbua, 2016-09-10:
If I remember correctly:
Ranging was TOF.
DW1000Ranging was mod TOF from decawave, see data sheet.

wazaahhh, 2016-09-14:
Thanks for the tip. I will investigate.

closed 2017-04-26 by Rotzbua

DrummyFloyd, 2017-06-30:
hi any news about the difference between them ?