thotro/arduino-dw1000 #119: Unique ID 0:0:0:0

meta0102, 2016-09-30:
just a question
if i run the basicconnectivity test the Unique ID return 0:0:0:0

DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: EUI: 0:0:0:0:255, OUI: 255:255:255
Network ID & Device Address: PAN: 10, Short Address: 5

now i'm not sure if i've destroyed the dwm1000 (mis wired...)

thanks

belforti, 2016-10-03:
If you see Device ID: DECA - model: 1, version: 3, revision: 0 at least it means that you are communicating to the module via SPI. If it were destroyed (I have one of those), you would not be able to read anything.

meta0102, 2016-10-04:
Ok, thanks. Hope you are right and i dont need the unique id for any communication / configuratiom to start ranging

closed 2017-04-24 by Rotzbua