thotro/arduino-dw1000 #121: Negative Distance

belforti, 2016-10-03:
Hi,

Thanks for sharing this project. I put it together using two Arduino Pro Mini 3.3v and get negative distances in the 4-digit order when the nodes are about 1.5mts away. I am powering the DWM1000 from the Arduino's Vcc pin since it can handle up to 150mA, and am not using any cap. Any debugging suggestion is greatly appreciated.

Thanks,
Gabe

jjviscomi, 2016-10-05:
This is most likely an overflow error.

belforti, 2016-10-05:
Thanks. Strange though since I'm using the examples from github and
typically you would get a sporadic overflow error, not a constant negative
distance. -G

jjviscomi, 2016-10-06:
Did you calibrate the device or just run it ?

belforti, 2016-10-10:
Joseph,

Thanks for your feedback. I ran it with 16384 (or 245ns) as TX and RX
antenna delay value, which is the default value. The problem is that the
system shows a distance of about -4500 mts, which means that even with the
full antenna delay value (65535) the system will never show a real distance
value. According to APS014 (page 6) the accuracy with no calibration is
30cms... not 4500 mts!

I did some testing adjusting the antenna delay values with another pair of
DWM1000 and was able to find a calibrated value that would allow me to work
within 40cms accuracy. However, after I saved the code and reset the
boards, the measurement were again completely off by several meters, even
if I left it running for a long time to stabilize temperature. Temperature
drift cannot account for so many meters according to the specs.I also added
some extra decoupling caps in case there were problems with the power
source, but nothing changed.

I can try to do the full calibration with three DWM1000s as explained in
APS014, however, I don't want to waste my time if this will not be the
reason for such an unstable measurement.

Best,
Gabe

jjviscomi, 2016-11-26:
@belforti I am curious if you resolved this? The APS014 specification of 30cm is for the EVK1000 specifically. Which example ranging code are you using, TWR or TDoA?

curlyz, 2016-11-26:
EVK can do TDoA?? @jjviscomi 
About antenna delay , I think every antenna module have difference delay ?
Have anyone got a method to calculate that delay individually for each DWM?