thotro/arduino-dw1000 #128: How Can I Do a TDoA Minimum System With this library?

maxicastle, 2016-11-03:
I'm trying to create a TDoA system using this library and 5 modules 1 Tag and 4 Anchors, but I don't understand how the library sends packets, my idea is send Blink messages from the Tag in regular intervals maybe 250 or 125ms in broadcast, and at same time make the ancors listen all the incoming messages and reseting the ToF Counter every 250 or 125ms (This part is not defined Wireless vs wired synchronization), any idea about how to send a brodcast message with this library?
Cheers.
[image]

curlyz, 2016-11-14:
Correct me if I am wrong , I have been researching this for month.
This library do TWR only . 
FYI : Pozyx system do ToA , LPS from Hackaday do TDoA and maybe AoA soon.

Rotzbua, 2016-11-15:
@curlyz This library is more a driver than an ranging system. It uses Roundtrip Time of Flight aka Two Way Ranging. It needs no sync, so it is easy to implement. I did not found another 
@maxicastle The dw1000 only sent raw packets, so you have to build the messages by your own. Only the `DW1000Ranging` example try to build and use well formatted packets.

Rotzbua, 2017-04-26:
Please move the information about TDoA to the wiki. Thanks.
duplicates #47 #77 #108 #128 #130 #152