thotro/arduino-dw1000 #129: PCBs ready

kuek, 2016-11-04:
Hi guys,
i created a PCB which takes 
an Arduino Pro Mini, DWM1000 chip and an ESP8266. 
It can be connected to 5V Mini USB, all 3V3 is handled by a voltage controller. The ESP is connected to the Arduino UART to send all ranging messages to Wifi to a server. This way, the anchors can easily be connected to a server and all ranging is done by the DWM1000. There is some additional resistors, capacitors, leds, connectors to flash the ESP etc.

they can be used either as anchor or as tag (depending on the ino you flash). let me know if you want some, you will get them at cost price.

[image]

az-z, 2016-11-04:
This is great work.
What if I do _not_ need WiFi to send messages?

kuek, 2016-11-04:
Hi az-z, 
no problem, just don't solder any ESP on the PCB. Its not needed at all...
I made an universal PCB, because I needed it WiFi for the anchors. But my tags also use the same PCB and don't use WiFi.

az-z, 2016-11-04:
Sweet.thank you

maxicastle, 2016-11-09:
How much? and where can i buy some of your boards.

kuek, 2016-11-09:
Hi, how many boards do you need?

maxicastle, 2016-11-11:
What about 10 boards.

kuek, 2016-11-11:
yep, no problem. each board is 4,33 eur cost price right now. let me know if you want some and drop me an email with the details.

curlyz, 2016-11-29:
@kuek Is this the board you use in your latest post ?

kuek, 2016-11-29:
Hi curlyz, 
yes, thats the boards I use. I also recoded these videos using 3 anchors and 1 tag of this hardware... 

https://www.youtube.com/watch?v=t1kuF8y7Lwg
and
https://www.youtube.com/watch?v=2MKSFAHrWqY

abencomo, 2016-11-29:
@kuek, how can I purchase 8 boards?  Can I send a private message so you can send me an invoice via Paypal?

faceless105, 2017-03-01:
I'd be interested in a few boards, are any still available?

kuek, 2017-03-02:
yep, still available.

sahilparekh, 2017-04-12:
Hi @kuek , I am also interested in four boards for now then post experimentation will like go go with more. Can you help me with only PCB boards, and all mounted boards?

kuek, 2017-04-12:
@sahilparekh yes i still have PCBs incl SMD. can you send me an email? kuek -at- me -dot- com

kuek, 2017-06-12:
Hi guys, we have a new version online which is based on an STM32F103 called Localino 2.0. You can still program the Localino 2.0 with Arduino IDE and make use of your own code. But it has much more space (128kB) and is heaps faster!!

vladimir-shmidt, 2018-02-23:
Hi @kuek, where can I find store? or maybe you can share your contact? is this board still available?

kuek, 2018-02-23:
Hi @vladimir-shmidt, yes we have some of Version 1.0 left, Please send us an email to info@localino.net

There are also newer boards available here www.localino.net/shop

vladimir-shmidt, 2018-02-23:
Hi,

Thanks. Version 2.0 board looks cool. There is some limitation on software around v2. Is it anchors count limitation only? Or anything more? 

I’m asking so as I’d like to buy couple kits and build playground with 5 people in with 10m2 space. I guess that it wouldn’t be enough to have only 4 anchors. So, will I able to program this shield by my own for example with this Arduino library https://github.com/thotro/arduino-dw1000 to avoid v2 software limitations?

Thanks.

From: kuek
Sent: 23 февраля 2018 г. 13:09
To: thotro/arduino-dw1000
Cc: Vladimir Shmidt; Mention
Subject: Re: [thotro/arduino-dw1000] PCBs ready - let me know if you want some (#129)

Hi @vladimir-shmidt, yes we have some of Version 1.0 left, Please send us an email to info@localino.net
There are also newer boards available here www.localino.net/shop

kuek, 2018-02-24:
Hi @vladimir-shmidt 

our v2.0 demo firmware can handle 4 anchors and 3 tags. 

But you can for sure program your own firmware on it. We supply the Arduino IDE which works with this Arduino https://github.com/thotro/arduino-dw1000 library, so you can develop your own code on our reliable hardware.
you will have full acccess to the STM32 controller, so you can do whatever you like ;-)

Alternatively you go for v3.0 which is finished and does not need development anymore.

escrowdis, 2019-04-12:
Hi Kuek,

This is really a nice work! I'd like to know if this PCB board is still available for academic research use? We bought couples of bb-dw1km and they didn't work fine in range with maximum distance of 30m under MODE_LONGDATA_RANGE_ACCURACY (we assume it could be at least 100m). 

Thanks!

Tony Weng

kuek, 2019-04-12:
Hi,

Yes we do have some PCBs left over.
In which country are you?

escrowdis, 2019-04-13:
Hi Kirk,

Thank you so much for the prompt reply. I’m in Taiwan. Is it available to
deliver? and how much does it cost for each? We may need 5 PCBs.

Thanks!

BRs,
Tony Weng

vanillabrand, 2019-04-13:
> issuecomment-482418845

I’d love some too if you have :)

NomisTuo, 2019-04-20:
Hello Kuek,

I am really interested by the work you did. Do you still have some PCB available? I would really like to test the precision of your solution! 

Thank you in advance for your reply.

Best regards

kuek, 2019-04-21:
Hi,

Yes we have some left over.
They are 65 EUR/piece net incl transceiver and all you need.

Regards
--
Heuel & Löher GmbH & Co KG
In den Höfen 9
57368 Lennestadt, Germany

website:  www.localino.net
linkedIn: www.linkedin.com/company/localino-rtls/

NomisTuo, 2019-04-21:
Hi Kuek,

If you still have enough PCB, I would be glad to get 6, i.e. 390€. I am living in Belgium (rue Bruyère 4, 1430 Rebecq) would it be possible to send them via post services? May I pay you over a bank transfer? 

I am maybe pushing to far (do not hesitate to tell me) but do you also have a schematic of the PCB with the different connexions and devices?

Thank you very much in advance.

Best regards.

Simon

kuek, 2019-04-21:
Hi Simon,

Please send me your details, address etc per email to info@localino.net

We will proceed then. Schematics are company confidential.

Best
--
Heuel & Löher GmbH & Co KG
In den Höfen 9
57368 Lennestadt, Germany

website:  www.localino.net
linkedIn: www.linkedin.com/company/localino-rtls/

NomisTuo, 2019-04-21:
Hi Kuek,

Thank you for your answer. I confirm my will to buy 6 PCB at 65€ each.

Here are my details:
Simon Verdebout 
Rue Bruyère 4
1430 Rebecq
Belgium
+32495465083
sim.verdebout@gmail.com

Best regards.

Simon

Yasirsmb, 2020-07-23:
> Hi Simon, Please send me your details, address etc per email to info@localino.net We will proceed then. Schematics are company confidential. Best -- Heuel & Löher GmbH & Co KG In den Höfen 9 57368 [...]

Hello, Can I do it with just using dwm1000 and Arduino mini pro without using PCB?