thotro/arduino-dw1000 #130: I think we should do TDoA instead of TWR

curlyz, 2016-11-14:
Here is my idea.
What if we use this lib for a drone , if my fuc'kin brother know about this library and setup a tag, it will interfere with my drone performance. In another word, sabotage.
And is it the Atmel in the Pro Mini sufficient to do this? Many other system seem to run on STM32 and I found out that it is just slightly more expensive.
Fix me if I am wrong 
@thotro

maxicastle, 2016-11-14:
Yeah TDoA is a superior technology than TWR, but is in fact more hard to develop, The Wired Synchronization system is better than the wireless synchronization system, I'm actually working in a Wired synchronization scheme for TDoA, using the next hardware:

- Operation mode One Shot Timebase Reset (OSTR) Mode

- CDCLVC1112 Fan-Out Clock Buffer

- AD8131ARMZ Differential amplifier (For drive signal in long distances)

-  Si5317 1–711 MHZ JITTER CLEANING CLOCK (not tested).

but i have a lot of work to do with this. the idea is create a CSU (Clock synchronization unit)

curlyz, 2016-11-15:
Have you try the crazyflie for reference ?

maxicastle, 2016-11-15:
No.

Rotzbua, 2016-11-15:
@curlyz There is a port of this library available (uses C instead of C++) which is used by a drone project. Maybe this is more useful for you.

curlyz, 2016-11-15:
@Rotzbua Can you be more specific about the drone project please :)

Rotzbua, 2016-11-15:
@curlyz see https://github.com/thotro/arduino-dw1000/issues/59#issuecomment-246112768

curlyz, 2016-11-17:
@maxicastle  How much will Wired synchronization scheme be in term of hardware may I ask ?.
Decawave have Wireless synchorinization developed and they charge for the code.

maxicastle, 2016-11-17:
Hi @curlyz 

Let me see if i can get your question:

Yes Decawave Offers a Wireless Synchronization scheme (Documentation, Code and Pseudo-codes), it's interesting, but is really expensive for a startup or some researcher for terms of confidentiality and Ethics I can not say the price here.

But in general terms of comparison a Wired System is more accurate and robust than a wireless system, or at least equal in accuracy in some cases.

The general view in hardware of a wired synchronization scheme is composed of the next units:

- CSU (Clock Synchronization Unit) a Black box that generate and distribute the sync signals around the anchors. some important blocks to start a design a CSE are the next:
x)The clock Generator MCU or IC
x)Fan-Out Clock Buffer
x)The Driver  to Drive the signal pulses in long distances without a big delay

-Connected Anchors, Anchors With the feature to connect to a local Network (RJ45, Wifi, CAN, Modbus etc)
-And a way to synchronize the Blink intervals of the TAGs, Slotted Aloha Can be the most viable way to do this.

I expect have resolve your doubt, by the way are you an investigator, student or are starting a project?

curlyz, 2016-11-23:
I am a student and want to implement this to my drone.
And btw , should the difference in the length between each anchor to CSU be considered too?

maxicastle, 2016-11-23:
yeah each distance from Anchor to CSU should be considered, because can have different Pd (Propagation delay), remember 3.3n seconds represent 1 Meter. the No consideration of this propagation delay for TDoA can generate accuracy problems, I suggest to read the about "Transmission Lines" only if you are not familiar with the topic, also decawave has a document talking about this "APS007 Wired Sync RTLS with the DW1000"  I left this file attached.
[APS007 Wired Sync RTLS with the DW1000.pdf](https://github.com/thotro/arduino-dw1000/files/609892/APS007.Wired.Sync.RTLS.with.the.DW1000.pdf)

curlyz, 2016-11-29:
@maxicastle Are you using DWM or DW ? 
DWM has no SYNC pin

maxicastle, 2016-11-29:
I'm actually using Both DW1000 and DWM1000, DWM1000 does not have Sync Pin Available, That Sucks!!, I'm don't know if its possible to hack the DWM1000 to get the Sync Pin.

BTW

I'm using DWM1000 for TAGS (ALOHA Blink) and DW1000 for TDoA Synchronized Anchors.

curlyz, 2016-11-29:
I am kinda stuck with DWM becuz I have bought dozen,
But I Think there is a work around, using about buffer clock to XTAL pin on atmegaso that all arduino run the same clock.and sync is by interrupt pin.
SYNC signal will be 2 kind of pulse, the first one will tell it to prepare , the second will reset timestamp, not timebase.
Also one thing, there is thermal drift effect, IDK if we can try to cool down it .

curlyz, 2016-11-29:
Dwm has a crystal ,so they disable any pin relate to it,XTAL,SYNC,TXCO

curlyz, 2016-11-29:
And I am not an arduino for you to interrupt folk 😂

curlyz, 2016-11-29:
Do you have adapter board for DW? I think we should have some talk

az-z, 2016-11-29:
I don't . Sorry.  I have a ton ( ok 7) dwm modules. And 2 out of 4 that I
soldered onto  https://oshpark.com/shared_projects/8M99bseq board refuse to
communicate(basicsender/basicreceiver).

maxicastle, 2016-11-30:
I'm Actually Working in a modification of this PCB From CIHOLAS http://www.ciholas.com/dwusb

curlyz, 2016-12-01:
so, no sync port also , is that what  you modify @maxicastle

curlyz, 2016-12-01:
My board get warm, but is it getTimestamp return 0 when no setTimestamp before it?

maxicastle, 2016-12-01:
that, the MCU the IMU and the External connection to Ethernet basicly.

Rotzbua, 2017-04-26:
Please move the information about TDoA to the wiki. Thanks.
duplicates #47 #77 #108 #128 #130 #152

werty37, 2017-12-26:
@maxicastle were you able to implement TDoA? I need some pointers as well for the hardware part.