thotro/arduino-dw1000 #132: Teensy RX not working

HappyDr0id, 2016-11-25:
Hello everyone,

I achieve to make Pro Mini 3.3V boards communicate and ranging between them quite easily. But I am facing an issue to do the same with Teensy 3.2 boards (which are also 3.3V).

I respect the testbed indicated in the Wiki (Teensy and Pro Mini have similar pins and the Teensy outputs 250 mA which is enough to power the DWM1000) and I manage to make the BasicConnectivityTest and BasicSender sketchs work.
With the BasicReceiver sketch, the interruption is raised but when the library is verifying if there is an error with the #isClockProblem() method, the CLKPLL_LL_BIT is set. If I artificially avoid this check, the message seems to be corrupted with unreadable special characters.

I saw that few persons have this same error with multiple devices based on the issue https://github.com/thotro/arduino-dw1000/issues/42 : Arduino Pro Mini as sender and Arduino Mega as receiver, Arduino Leonardo as sender and receiver, Arduino Uno as sender and Arduino Ethernet as receiver. 
This is perfectly working for me with Pro Mini 3.3V boards, so this is certainly linked to a material issue (the fact that the processor on the Teensy 3.2 is not an ATMEL 328 ?).

Does anyone actually use some Teensy boards to communicate with DW1000 chips ? 
Thanks in advance for any help about this subject.

Rotzbua, 2016-11-25:
Did you try it with patch #131?

HappyDr0id, 2016-11-28:
Thanks to the last merge of the #131 patch, I am now able to send an receive messages with Teensy. But please note that I had to configure the IRQ pin as INPUT to make it work. I added this code in the ::begin function of the DW1000 class :

```
void DW1000Class::begin(uint8_t irq, uint8_t rst) {
    // generous initial init/wake-up-idle delay
    delay(5);
    // Configure the IRQ pin as INPUT
    pinMode(irq, INPUT);
    // start SPI
    SPI.begin();
```

Some clock errors are still raised time to time but I suspect that this is due to a wire on the breadboard so I will investigate on this point and complete my answer as soon as I have more informations.

Rotzbua, 2016-11-28:
I think the different implementations of the HAL cause the need of a `pinMode(irq, INPUT);`.
But I see no reason against an additional `pinMode(irq, INPUT);`, so may make a pull request? @HappyDr0id

HappyDr0id, 2016-11-28:
The communication between Arduino Pro Minis is also working even if I add this pinMode configuration, so yes, I will make a pull request to add this and fix the Teensy problem.

cimbalo, 2016-12-31:
I solved a similar issue on ESP8266 by changing the SPI max speed.
https://github.com/thotro/arduino-dw1000/blob/master/src/DW1000.cpp#L99
18Mhz seems an appropiate value with Teensy running at 72Mhz (SPI_CLOCK_DIV = 4).

sarwadenj, 2017-01-12:
@cimbalo hi I'm trying DWM1000 with ESP12E. I'm able to run basic connectivity program but unable to do basic sender and receiver programs.  How you made setup and what changes you made.

cimbalo, 2017-01-12:
@sarwadenj as I wrote before I solved the issue by changing SPI max speed.
https://github.com/thotro/arduino-dw1000/blob/master/src/DW1000.cpp#L99
The value I used for ESP8266 is 20Mhz.

Rotzbua, 2017-04-24:
@cimbalo Please add your hint to the wiki. Thanks :)

Fuhu-che, 2019-12-27:
> Unfortunately I still get a lot of clock errors and they are not due to a bad contact or wires (the same breadboard and cables are working perfectly with Pro Mini boards). I think that there is still another implementation problem than the LDE load.
> Futhermore I have to overclock my Teensy boards to 96Mhz (normal is 72Mhz) to have some working communications which is weird. I will investigate on the Decawave configuration part and update the issue if I get any details.

Hi Bro, 
I am also working on this project as well. Can I ask have you work it out now?Thanks.