thotro/arduino-dw1000 #133: Demo Videos, python processes in realtime

kuek, 2016-11-29:
hi,
just wanted to share these two videos we recorded. behind the positioning is a python processing...

https://youtu.be/t1kuF8y7Lwg

https://youtu.be/2MKSFAHrWqY

[image]
[image]

jack880019, 2016-12-07:
Hello good works for u guys~
I am working on this kind of project too. Now I encounter some problem wish you can give me some advise. I am using the RangingAnchor example code to calculate the range. However  I find that when my Tag connect to the first anchor. it will continously calculate the range which is very good for me. but after I connect second anchor. the range calculation of the first Anchor start to have some time delay which I don't want. Can U give me some advise Thanks!!!!!

Rotzbua, 2017-04-24:
@kuek Hi, can you add this to our wiki? Would be nice :) . Thanks!

kuek, 2017-06-12:
@Rotzbua sure I will do. There is also an update of the firmware v0.4a online, which supports to locate multiple tags inside the room. I have also added some tracking and filtering.... Kalman is still on the list. but I will also add that!