thotro/arduino-dw1000 #135: Calibration of Reply Delay Time

berkerbatur, 2016-12-05:
Hi,

I'm testing RangingAnchor & RangingTag examples and I wonder do we have to calibrate replyDelayTimeUS, before start ranging protocol using a pre-defined range? Or the value is not important as long as we do not get HPDWARN event? 

With default 3000 given, I got large distance errors, and after setting it to 5000, I  reached ~10 / 20 cm. range error. I'm using 3.3V arduino pro mini, dwm1000 adapter board and basic breadboard with standard jumper cables to connect them, could it be reason that 3ms cause some range drifts, maybe triggering HPDWARN event ?

As I understood, this delay must be same on both side in Symmetric Double Sided, is this also valid for Asymmetric one ? I couldn't get it clear from comment of the variable. 

Best regards,
Berker

grex93, 2017-01-21:
i have the same problem, do you solve it?

berkerbatur, 2017-01-23:
Hi, I haven't implemented calibration method for replay delay time yet. In TWR (RangingAnchor/Tag) I simply get best accuracy with  5 milliseconds. But it depends of several things, whether you iterate through anchors using round-robin or use dynamic anchor discovery with blinks or implement another technique to divide anchors/tags to time slots.

If you use DWM1000Ranging_Anchor/Tag examples, you have a chance to exceed timer delay depending on your network devices count, see the calculation block:

`_timerDelay = DEFAULT_TIMER_DELAY+(uint16_t)(_networkDevicesNumber*3*DEFAULT_REPLY_DELAY_TIME/1000);`

hdineth, 2017-10-31:
hello,
Have you find a proper way to change the delay.