thotro/arduino-dw1000 #136: Device ID: 00 - model: 0, version: 0, revision: 0

yigibihe, 2016-12-06:
Hello,

I have been trying to get ranging example to work for a week. When I run the connectivity test it usually returns this:
`Device ID: 00 - model: 0, version: 0, revision: 0`

But occasionally it returns `Device ID: DECA`, in which case ranging works fine.

I have connected my module as the example test board, added an 330 ohm resistor for IRQ pulldown. 4 lines of SPI, IRQ and reset are connected. I'm using a 3.3 v arduino mini pro to power and drive the module. I also added 500uF capacitor to the power line of dvm1000 for power level stability. I have few dvm1000's , they seem to be working whenever they want. I change the module without changing the wiring, still the same modules sometimes work and sometimes does not.

Rotzbua, 2016-12-06:
I have almost the same config. Sometimes I get no output or corrupted. The `ATMEGA32U4` need some startup time.
My workaround: Add after `Serial.begin(9600);` a `while (!Serial);` or a `delay(5000);`.

yigibihe, 2016-12-12:
Solved: PCB's were bad. New PCB's solved the problem. I also added the while to setup for extra stability. Thank you.

az-z, 2016-12-12:
What pcbs did you use?

yigibihe, 2016-12-12:
I got the pcb in the repo printed. First guy screwed up at printing, second guy did a better job.

az-z, 2016-12-12:
And how did you figure out that it is a pcb?

yigibihe, 2016-12-12:
Multimeter buzzer found out the shorts. Still pretty ashamed for not checking them when I first got them.

grex93, 2017-01-07:
hi, 
i'm using an arduino mini pro, and i attach pin like in the example ( photo) .
I have for the basic test this result:
1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 0, version: 0, revision: 0
Unique ID: 00:00:00:00:00:00:00:00
Network ID & Device Address: PAN: 00, Short Address: 00
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

is it correct? because i have id: DECA but the others info they are all zero...

Rotzbua, 2017-01-07:
@grex93 That's wrong. `model: 0, version: 0, revision: 0` should be non zero.

hmatulis, 2017-01-07:
[image]
Apparently you have the blue and yellow cables connected to ground and nothing connected to MOSI and CS.  Double check your wiring.

grex93, 2017-01-07:
i follow exactly this image...
[image]
i'm sure about the red one and the blu one ( VDDAON and GND) , but for the others i follow the image .. 
i can't find any other image of the connection with arduino

az-z, 2017-01-07:
I have a really bad experience with soldering; )

grex93, 2017-01-07:
hmatulis  can you help me with the pin connections?

hmatulis, 2017-01-08:
Sure grex93, 
The fritzing image refers to the pcb , not to the DWM1000 itself. I suggest you to download the DWM1000 datasheet from decawave site:  http://www.decawave.com/sites/default/files/resources/dwm1000-datasheet-v1.3.pdf 
In page 9 you have the DWM1000 pin assigments (the same image I posted) 
Assuming that you are using an Arduino Pro Mini 3.3 Volts (The DWM1000 works at 3.3V!!!!)
// connection pins (arduino code)
const uint8_t PIN_RST = 9; // reset pin
const uint8_t PIN_IRQ = 2; // irq pin
const uint8_t PIN_SS = SS; // spi select pin
  DW1000.begin(PIN_IRQ, PIN_RST);
  DW1000.select(PIN_SS);

You should connect:
DWM1000                                                      Arduino Pro Mini 
Pin 24 or Pin 23 or Pin 16 or Pin 8     to       Ground
Pin 22 IRQ/GPIO8                                          PIN_IRQ ( pin 2) 
Pin 20 SPICLK                                                SCK (pin 13)
Pin 19 SPIMISO                                              MISO (pin 12)
Pin 18 SPIMOSI                                              MOSI (pin 11)
Pin 17 SPICSn                                                PIN_SS (pin 13) 
Pin 6 and/or Pin 7 VDD3V3                           Arduino Vcc (3v3)   
Pin5 VDDAON                                                Arduino Vcc (3v3)   

Please double check with the Decawave document.....

I hope it helps

grex93, 2017-01-08:
ok, so i must connect spiclock and spicsn ( 20,17) both to the 13° on arduino pro mini? 
the resistor i used for connect the irq, will i need it, is it for another connection,  or  can i  delete it? 

thanks a lot!

grex93, 2017-01-08:
Now i have:
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

i think it's working!!!! thanks a lot...

last question, which I saw in this thread (https://github.com/thotro/arduino-dw1000/issues/58) connecting an Arduino Arduino Nano and 2 (both with 5 volts pin) to the module ..
which is the risk? Can I connect the module to a nano / Arduino Uno with some resistance for each pin (i don't know, however, what resistors used)?

hmatulis, 2017-01-08:
You also need a pull-up resistor between DWM1000 VSS (Ground) and DWM1000 IRQ (pin 22)

Regarding using 3.3v logic with 5v circuits, it may work , but it will probably fry in the medium/long run. As this is not a cheap module, I'll definitely advise to use some sort of voltage translation.
Adafruit has a useful explanation of level shifting - https://learn.sparkfun.com/tutorials/voltage-dividers
[image]
using Vin=5 and Vout=3.3   
R2 must have twice the resistance of R1

They recommend resistors between 1k and 10k ,
So R1 = 1k and R2= 2k or
R1=3.3k r2=6.6k or
R1 = 1.8k   R2 = 3.3k

You can also buy or build a level shifter :  https://www.sparkfun.com/products/12009

http://www.ebay.ca/itm/8-channel-Bi-Directional-Logic-Level-Shifter-Converter-3-3V-5V-For-Arduino-/391148663709?hash=item5b1246f39d:g:VDoAAOSwBahVWrIB

grex93, 2017-01-08:
thanks  a lot! =)

Rotzbua, 2017-04-24:
@hmatulis Hi, please can you add your post to the wiki? Thanks! :)

Ruben-26, 2017-12-23:
In my case what fails me is that I do not have the level shifter?
[image]

Ruben-26, 2017-12-23:
[image]

DricSoh, 2018-03-02:
Hi, I got similar problems.
Device ID: 00 - model: 0, version: 0, revision: 0
Unique ID: 00:00:00:DE:CA:01:30:FF
Network ID & Device Address: PAN: CA01, Short Address: 307F
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Please advise me :(

Rotzbua, 2018-03-02:
@DricSoh Read the complete thread and some others and write your summary into the wiki. This project is open source and a lot of informaiton is available. Help your self or consult somebody, but then you have to pay...

fluentart, 2019-03-31:
Refer to this diagram for pinouts if following the fritzing image: https://github.com/thotro/arduino-dw1000/blob/master/adapterBoard/adapterBoard%20pdf/adapterBoard_etch_copper_top.pdf