thotro/arduino-dw1000 #137: Range measurement slow down when connect to multiple Anchor

jack880019, 2016-12-07:
Hello guys~
I am working on this kind of project too. Now I encounter some problem wish you can give me some advise. I am using the RangingAnchor example code to calculate the range. However I find that when my Tag connect to the first anchor. it will continously calculate the range which is very good for me. but after I connect second anchor. the range calculation of the first Anchor start to have some time delay which I don't want. And I also find that in this condition the second Anchor can measure the range continously. Plz give me some advise Thanks!!!!!
best regards

az-z, 2016-12-18:
> .. the range calculation of the first Anchor start to have some time delay 
will be helpful to have exact degradation metrics.

berkerbatur, 2017-01-29:
@jack880019 you should get similar sampling rates for each anchor if all of your components use same components and power sources, but decrease in sampling rate when you increase anchors is normal since tag starts time slotting each anchor

closed 2018-08-30 by Rotzbua