thotro/arduino-dw1000 #138: "Not found" in anchor example

yigibihe, 2016-12-12:
Hello, when I upload the anchor example I keep getting these messages. I uncommented the extra debug info from source.

`Not found
unknown: 8D:FE
Not found
unknown: 8D:FE
`
Digits change with almost every reboot.

Rotzbua, 2016-12-12:
@yigibihe there are more than one anchor example, which is it?

yigibihe, 2016-12-12:
"DW1000Ranging_ANCHOR" example.

sarwadenj, 2017-01-12:
@yigibihe I'm too facing same problem,you get the data or not?

Slamoth, 2017-02-16:
How many anchors and tags are you trying to make it work with ?

hu265, 2017-03-03:
I got the same results from that example and I'm testing 1 tag with 1 anchor.

khawajamechatronics, 2017-03-06:
I have same issue, working fine 1-1 communication but as soon as I add more tags or anchors. its give not found frequently

hu265, 2017-03-21:
@khawajamechatronics Did you change anything on the code? What does your result look like for the 1-1 communication? Mine keeps on giving me "blink" and "delete inactive device".

closed 2018-08-30 by Rotzbua