thotro/arduino-dw1000 #140: API docs are broken

az-z, 2016-12-18:
Gents,
 it appears that [API html docs are empty](https://cdn.rawgit.com/thotro/arduino-dw1000/master/extras/doc/html/index.html). Is there a way to fix this?

Rotzbua, 2016-12-18:
Doc is old, but not empty:
https://cdn.rawgit.com/thotro/arduino-dw1000/master/extras/doc/html/annotated.html

az-z, 2016-12-18:
> doc/html/annotated.html

Sorry I thought api should refer to methods/functions and "data structures
"to data structures.
Does "old" means outdated=invalid? Or it (doc) is regenerate automatically
somehow?

Rotzbua, 2016-12-19:
> Or it (doc) is regenerate automatically somehow?

no, no continues delivery available

> Does "old" means outdated=invalid?

old means (time) old... outdated for sure
` Generated on Sun Jul 12 2015 00:36:00 for Arduino driver library for Decawave DW1000 by   doxygen 1.8.6 `

az-z, 2016-12-20:
Would you be interested to setup something like [this](http://rickfoosusa.blogspot.com/2011/10/howto-use-doxygen-with-github.html)

closed 2017-01-04 by thotro