thotro/arduino-dw1000 #143: It seems incorrect to adjust the range bias caused by received signal power.

wql7013, 2016-12-28:
According to Page 12 of application note aps011, the reported distance can then be corrected such that: Actual distance = Reported distance - Range Bias Correction. But in the correctTimestamp method of DW1000Class, it is "Actual Timestamps = Reported Timestamps + Range Bias Timestamps" rather than minus.
Furthermore, why did you use A = 115.72 when estimating the signal power, while it is recommended A = 113.77 as per section 4.7.2 – Estimating the receive signal power in the DW1000 user manual ?
Besides, according to application note aps011, the received signal power is calculated by Friis' path loss formula rather than the estimated power calculated by register file value. Is it accurate to adjust the range by the estimated power ?

wql7013, 2017-02-18:
Is there anyone else finding these issues?

StevenH2812, 2017-06-20:
It seems like you are correct on both points. According to the documentation provided by decawave the sign in correctTimestamp should indeed be negative. I also don't see a reason why A = 115.72, since the documentation says it should be 113.77. It is especially inexplicable because the A value for the 64 MHz PRF case does match the value in the documentation.

When you look at the documentation though, the APS011 actually gives a different equation to estimate the power of the unobstructed first path. In this document they use this power of the unobstructed first path in the lookup table for the range bias.  Then in the code in order to get the receive power the equation from the DW1000 user manual is used which gives the received signal power (of all the arriving paths). Perhaps this won't give correct results since the lookup table requires the power of the unobstructed first path and not of all the arriving paths.

I started looking into this because I am getting some strange ranging results. For me the ranging bias is dependent on the range itself. Basically I can calibrate the modules at a certain distance (by changing the antenna delay) and they can give the correct range at this distance, but when I try to measure a different distance, it will show a range bias that is equal to how much I deviate from the calibration point. So say I calibrate it at 1 meter distance, and then try to measure at 1.5 meters, the modules will report 1.6 meters (deviation +0.5 meters, range error +0.1 meters). Then when I measure at 0.5 meters the modules will report 0.4 meters (deviation -0.5 meters, range error -0.1 meters). Are you experiencing similar issues, or do you have another reason to look into this?

curlyz, 2019-02-21:
I am wondering why this is not a PR :(