thotro/arduino-dw1000 #145: Get started on Arduino Pro Micro

Nyan-Kat, 2017-01-01:
Hello (and happy new year everybody),

We're currently struggling with our Arduino Pro Micro. We decided to use them instead of Arduino Pro Mini as it's the case in all the other issues. As a result, the SPI frequency is 8 MHz instead of the usual 16 MHz. Thought, the output tension of all ports is 3,3V. 

We first powered directly the DWM1000 with the Pro Micro but we read that 160 mA we necessary to power the chip. We changed for a Leonardo to power the DWM1000 (3,3V - 260 mA) and only for power. We tested ranging programs at first but they didn't produced any result, so we changed for a simpler one : BasicConnectivityTest.ino. And the only results we get is : 

```
Device ID: 00 - model: 0, version: 0, revision: 0
Unique ID: 00:00:00:00:00:00:00:00
Network ID & Device Address: PAN: 00, Short Address: 00
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
```

We tried to change the setup a few times but the result is still the same. Note that is the default result we get even when we don't connect the DWM1000 to the Pro Micro.

Do you think the problem comes from the frequency of the Pro Micro (8MHz), or any other idea ?

Thanks in advance for your help.

Rotzbua, 2017-01-02:
> We first powered directly the DWM1000 with the Pro Micro but we read that 160 mA we necessary to power the chip. 

Why you changed the power source? The data sheet mention 3,3V at 500mA current. So it should be possible to power the dw1000 with only the pro micro.

See https://cdn.sparkfun.com/datasheets/Dev/Arduino/Boards/ProMicro8MHzv1.pdf

frankdabank, 2017-01-04:
You may need to change the interrupt pin in the library.
https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

Nyan-Kat, 2017-04-20:
Hi, we solved the problem, it was caused in the program by the following line. 
`const uint8_t PIN_SS = SS; `
On the Arduino mini, the SS port is note define on the Arduino Pro Micro. To solve the problem, you only have to replace the following line to define the PIN_SS. 
`const uint8_t PIN_SS = 10; `

closed 2017-04-20 by Nyan-Kat