thotro/arduino-dw1000 #146: Using this code with TREK100 modules

savior-help, 2017-01-03:
hi
my question is can i use this code as tag along with decawave TREK1000 modules as anchors?
which configuration should i make or which example can i use.
i really appreciate your help.
thanks

Rotzbua, 2017-01-03:
Seems to be similar to #61 .

savior-help, 2017-01-03:
so what was the conclusion there????
i didn't see any certain answer there?
does this code support multi tags along with for example 4 anchors in general???

Rotzbua, 2017-01-03:
Wow so much question marks. 

As you can read in the `readme`:
> A library that offers functionality to use Decawave's DW1000 chips/modules with Arduino 

Arduino != TREK100 ==> high chance that it will not work.

cimbalo, 2017-01-12:
You can still disable TREK1000 ARM and connect the DW1000 chip to an Arduino using the TREK1000 SPI header (J6).

Please refer to the TREK1000 manual for the correct procedure.

closed 2017-04-24 by Rotzbua

brettdaman, 2018-01-23:
Technically it should be possible, I also want to do this but even with the correct mode settings no ranging data is received.