thotro/arduino-dw1000 #147: Could Tag Compute Range?

fabio-C, 2017-01-06:
Can I have a system with 3 fixed Anchor, and 1 mobile Tag, in wich the Tag compute sequentially the calculation of distance between each anchor, and then perform trilateration?

In the code example I noted that only Anchor compute the range, and Tag only polls for range computation.

berkerbatur, 2017-01-06:
Possible. I guess you only need to add _timeRangeReceived_ to data and send it back to tag without any twr computation, when anchor receives _RANGE_ message.

fabio-C, 2017-01-06:
I noticed that in the DW1000Ranging_Tag / DW1000Ranging_Anchor example code, both Tag and Anchor have the function "newRange()" that can print the computed range. So both Tag and Anchor can output the range?
I am a bit confused.

berkerbatur, 2017-01-06:
Yes you can output the range in both of them. 
In both, anchor computed the range and tag gets it with RANGE_REPORT message.

fabio-C, 2017-01-06:
Ah ok.
Thank you!

closed 2017-04-26 by Rotzbua