thotro/arduino-dw1000 #148: Basic Sender doesn't work

grex93, 2017-01-08:
Hi, 
I'm finally have two dwm working, with two arduino pro mini and 10k resistor for the ground and the irq pin. 
I first of all tested the basicconnectivity with good results.

But i have problems with basicsender/receiver, in particular with the receiver. 
Testing the sender i have:
Transmitting packet ... #0
ARDUINO delay sent [ms] ... 1418
Processed packet ... #0
Sent timestamp ... 298895.50
DW1000 delta send time [ms] ... 298.90
Transmitting packet ... #1
...ECC...

Testing the receiver i obtain:
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 06
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Error receiving a message
Error data is ... 

I tested both module with both code, starting before one, and another time another one.. 
I check my connection i think they are correct, the pin in the code are the  same..
22 IRQ- 10k resistor-  2 of arduino 
24 VSS-10k resistor-  ground 
20 spiclk - 13 spiclock
19 miso - 12 miso
18 mosi - 11 mosi
17 spicsn - pin ss 10 
3 rstn -  9 
vddaon , vdd3v -   vcc

can you help me?

hmatulis, 2017-01-08:
This configuration works for me:
[image]

grex93, 2017-01-08:
It is the same, i am using an arduino pro mini . This is only 3.3volt and a normal ftdi serial converter. Maybe the serial converter of 5v? I am using one serial adapter with only 3.3v ( born  for arduino pro mini) and one universal with vcc 5v. But this is connect to arduino...

But the sending i think is correct. Why this? Why The riceiving interrupts seem to don't work?

hmatulis, 2017-01-09:
I'd suggest to measure the voltage btw vcc and gnd. 
Excessive voltage can be the source of the problem 
Some FTDIs have a 3.3v pin hole

sarwadenj, 2017-01-09:
basic sender --->

Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: 00:00:00:00:DE:CA:01:30
Network ID & Device Address: PAN: DECA, Short Address: 130
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
ARDUINO delay sent [ms] ... 312
Processed packet ... #0
Sent timestamp ... 228.50
DW1000 delta send time [ms] .

basic receiver----->

Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: 00:00:00:00:DE:CA:01:30
Network ID & Device Address: PAN: DECA, Short Address: 130
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5

on receiver side message not getting......

grex93, 2017-01-14:
I finally managed to solve the problem! pin 2 dell'arduino had no copper in place. After many attempts I changed pin using pin 3 dell'arduino interrupt and now it works!

I tried both the range codes (dwm1000 and the other) noting an error of about one meter and a half from any distance. the antenna has no batteries or other small storage or metallic objects.
how can I calibrate or improve distance (through a still and a tag)?

closed 2017-04-24 by Rotzbua

dhaqqi, 2022-10-25:
Hi im facing the same problem here, i already check all the hardware and it is just like the one everyone post but i stull can't manage to make basic sender and basic receiver work

im really confused and have been working for this for months but still stuck in the same problem. program just stop at 

[image]

an its never go into loop phase