thotro/arduino-dw1000 #149: Problem with 3 anchors

fabio-C, 2017-01-09:
I was able to manage 2 Anchors and 1 Tag using the sample code DW1000_RangingTag/Anchor.
The tag printout the distance compared to each Anchors.

But when I power-on a third Anchor, I have an issue.

In particular the print say continuously:
```
ranging init; 1 device added ! ->  short:4300
delete inactive device: 4300
ranging init; 1 device added ! ->  short:4300
delete inactive device: 4300
ranging init; 1 device added ! ->  short:4300
delete inactive device: 4300
```
and it stop to range the other two.
The problem is not the particular tag, because it happen with every combination of two tag, adding the third.

What could be the problem?

My setup is:
Arduino Pro Mini (3.3V, 8MHz), Decawave DMW1000. All tags and anchors are battery powered with two triple A battery (it is enough?).
I try to changed the mode (MODE_LONGDATA_FAST_ACCURACY, MODE_SHORTDATA_FAST_ACCURACY, etc...) but the problem continue.

fabio-C, 2017-01-09:
Solved using Arduino 1.6.8.
As mentioned here: https://github.com/thotro/arduino-dw1000/issues/116

berkerbatur, 2017-01-09:
Hi fabio,

We faced exactly same problem in DW1000_RangingTag/Anchor example, and couldn't solve it yet. Our setup is similar, we power the circuit using USB power supply. 

We suspected giving different reply delay times to each anchor (DW1000RangingClass::transmitPoll method), configured reply delay times but still unable to get a working system.

We updated RangingAnchor/Tag example and got a working system with 3 Anchors and a Tag. We don't use Blink phase, and hard-coded all anchor addresses and tag's address.

berkerbatur, 2017-01-11:
Hi,
has anyone manage to get a working DW1000_RangingTag/Anchor example with 3 Anchors, using Arduino pro-mini 3.3V 8MHz, using Arduino 1.8.1 version?

zohaib-arshad, 2017-01-12:
Hello @fabio-C .
We have the same issue. How did you exactly resolve it?

Did you just program all your anchors and the tag by the Arduino Version 1.6.8 and it worked?

fabio-C, 2017-01-12:
@zohaib-arshad Yes.

sarwadenj, 2017-01-12:
Hi @fabio-C, 
[image]
  
you did set up like this?
Also have you made any changes in code or library?

fabio-C, 2017-01-12:
Yes this setup.
I used the DW1000_RangingTag/Anchor examples. I changed only the ranging mode (MODE_LONGDATA_RANGE_ACCURACY).
External power supply. Arduino 1.6.8. Mac OS.

zohaib-arshad, 2017-01-12:
We are also having some power issues.

We are using 4 DWM1000 Modules and Powering them using external DC 3.3 V Power Supplies.
The Arduino Pro Minis 3.3 V are being powered using the USB to TTL Converters by my Laptop.

But the Voltage at the DWM modules falls below 3.3 V to around 2.7 V.

Can this also be an issue for my multi-anchors connection failure?

sarwadenj, 2017-01-12:
Hi @fabio-C,
when I'm using Arduino 1.6.8 i'm getting this error
In file included from C:\Users\vishal\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Time.h:38:0,

                 from C:\Users\vishal\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000.h:37,

                 from C:\Users\vishal\Documents\Arduino\libraries\arduino-dw1000-master\src/DW1000Ranging.h:29,

                 from C:\Users\vishal\Documents\Arduino\libraries\arduino-dw1000-master\examples\DW1000Ranging_ANCHOR\DW1000Ranging_ANCHOR.ino:9:

C:\Users\vishal\Documents\Arduino\libraries\arduino-dw1000-master\src/require_cpp11.h:9:2: error: #error "This library needs at least a C++11 compliant compiler, maybe compiler argument for C++11 support is missing or if you use Arduino IDE upgrade to version >=1.6.6"

 #error "This library needs at least a C++11 compliant compiler, maybe compiler argument for C++11 support is missing or if you use Arduino IDE upgrade to version >=1.6.6"

  ^

Using library SPI at version 1.0 in folder: C:\Users\vishal\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.8\libraries\SPI 
Using library arduino-dw1000-master at version 0.6 in folder: C:\Users\vishal\Documents\Arduino\libraries\arduino-dw1000-master 
exit status 1
Error compiling for board Arduino Pro or Pro Mini.

fabio-C, 2017-01-12:
@zohaib-arshad Yes it could be. I suggest you to power the tag with an external supply in addition to usb. I noticed that when there is a power supply issue  the Arduino reset itself and the program restart, so you can see it in the Serial output.

@sarwadenj I had the same problem first time because I had a old version of Arduino. Then I updated to last Arduino and this error disappear. (Strangely) when I downgraded to 1.6.8 this error doesn't happears again.

AliSiddiqui4, 2017-01-13:
I am getting this error after uninstalling my arduino 1.6.13 and then installing 1.6.8.

avrdude: verification error, first mismatch at byte 0x0002
0x8b != 0x8e
avrdude: verification error; content mismatch

rbrtwtrs, 2017-03-22:
Does anyone have a sure fix for getting three anchors to work? I can run any two at a time but the third causes it to no longer get any ranges.

AliSiddiqui4, 2017-03-22:
@rbrtwtrs use other modes, the 3rd anchor doesn't work with the default mode. For me the "LONGDATA_FAST_LOWPOWER" works with 3 anchors.

hu265, 2017-03-22:
Hi @AliSiddiqui4 
So did you end up using 1.6.8 or 1.6.13? And is the "LONGDATA_FAST_LOWPOWER" mode going to work for the 1 tag 1 anchor case?

AliSiddiqui4, 2017-03-22:
The ardunio version doesn't matter, and yes it should.

rbrtwtrs, 2017-03-22:
Great, I will give it a try.

Rotzbua, 2017-04-26:
solved - according to first comment

closed 2017-04-26 by Rotzbua