thotro/arduino-dw1000 #150: basic connectivity test getting Device ID: FFFF - model: 255, version: 15, revision: 15

sarwadenj, 2017-01-10:
Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: FFFF - model: 255, version: 15, revision: 15

Is Device ID FFFF is right.

Rotzbua, 2017-01-10:
No.

sarwadenj, 2017-01-11:
@Rotzbua sometimes I'm getting Device ID as DEC8, 7F7F too so should i think for DWM1000 DECA is constant .

closed 2017-01-12 by sarwadenj

Rotzbua, 2017-01-12:
@sarwadenj It means your connection between controller and dw1000 is crap -> random bitflips.

werty37, 2017-09-08:
Hi,

I am facing this same issue. But i had printed Wayne Holders TWR board from Oshpark. I am using a pro mini 5v. 

It must be something to do with the pins. I have not been able to get a schematic / circuit diagram for the Wayne Holder's TWR board. Please guide.

```
// connection pins
const uint8_t PIN_RST = 9; // reset pin
const uint8_t PIN_IRQ = 2; // irq pin
const uint8_t PIN_SS = SS; // spi select pin
```

Thanks
S