thotro/arduino-dw1000 #151: what should be device id?   basic sender not working check what's coming ...

sarwadenj, 2017-01-12:
after running basic sender code o/p------>

### DW1000-arduino-sender-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Transmitting packet ... #0

grex93, 2017-01-16:
in device ID you must have "DECA" for example. 
I also have had this problem. One possible cause is the wrong link or an imperfect welding.

gsgoraya, 2017-01-18:
I am facing the same issue. So far, I have figured that the IRQ (interrupt) is never generated by the dwm1000. The IRQ pin stays LOW all the time. I was expecting the DWM1000 to generate an interrupt when it is done transmitting.

sarwadenj, 2017-01-18:
@gsgoraya have you can check it by printing bit in system event status register.

gsgoraya, 2017-01-18:
@sarwadenj , thanks for your reply. I am very new to the embedded programming. I will have to figure out how to print bit from 0x0f :) I am searching on that topic. Also if you can tell me how to do that, I will be thankful.

Kind Regards

gsgoraya, 2017-01-18:
@sarwadenj and btw, I checked the IRQ pin using a multimeter. It shows Grounded all the time. If I connect it to a 3.~ volts using a jumper wire, the interrupt is generated and the programming loop continues.

gsgoraya, 2017-01-18:
ok, I was able to print the required bits. 
TXFRS i.e., 7 on 0x0F, it is 0 in the beginning, but turns to 1 when a packet is sent. Which is the correct behavior.
MTXFRS i.e., 7 on 0x0E, is also set to 1, all the time.

So, the IRQ pin needs to do some activity, as soon as the TXFRS turns to 1 from 0. But for some reason unknown (so far), it does not do that

sarwadenj, 2017-01-18:
Great..
Ya as we call  DW1000.newTransmit(); TXFRS i.e., 7 becomes 1.

sorry i didn't get your point--->But for some reason unknown (so far), it does not do that.

just have a look at user_manual 85 page.

gsgoraya, 2017-01-18:
@sarwadenj .. Hi, I am away from the setup now, so cannot tell you the device ID. I will post it as soon as I get back.

"But for some reason unknown (so far), it does not do that" <- I meant that, the IRQ pin on the dw1000, is required to generate interrupt when the TXFRS switches from 0 to 1, but it is not generating at interrupt in my setup.

I was going through the code, and noticed that in DW1000.cpp, it has setInterruptPolarity(true);
This sets the bit HIRQ_POL to set IRQ to active-high ( I am assuming here that active-high means, that the pin stays at positive 3.3 v when there is no interrupt) (or it could mean the opposite, not sure)

Ok, in case, active-high means that the default state of IRQ is to stay at 3.3 positive. And an interrupt would cause the pin to go LOW. Well, then attachInterrupt(digitalPinToInterrupt(_irq), DW1000Class::handleInterrupt, RISING); in DW1000.cpp is expecting the interrupt to be from low to high. Which appears contradictory. (I will have to test that). Ok, btw, I do not mean to confuse anyone with what I have written here. I am a beginner at these things :) just trying to make sense of things.

gsgoraya, 2017-01-18:
@sarwadenj , here is the output I get

> ### DW1000-arduino-sender-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 110 kb/s, PRF: 64 MHz, Preamble: 2048 symbols (code #10), Channel: #5
Transmitting packet ... #0

sarwadenj, 2017-01-19:
@gsgoraya  make sure your connections are right having 10k or less than 10k resistor connected to IRQ.
Device ID should be DECA . For first time i also got this error but after checking Hardware it started.
Wait i will also checks ones. If you got something let me know.

gsgoraya, 2017-01-22:
@sarwadenj Hi mate, just tested out the connections. Everything is in order. no disconnects, no shorts. I do have a 10k resister on the IRQ. btw, I am following this setup
https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

I will post, if I find some clue. Regards

sarwadenj, 2017-01-22:
[image]
@gsgoraya DECA should be the device id . Make sure your dwm1000 should get appropriates voltage.

sarwadenj, 2017-01-25:
@gsgoraya as you are using foolwing setup have you made changes
https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

----->
I'm using to drive the DWM1000 module was designed to work with the Arduino Pro Mini and hard coded to use interrupt 0 which, in the Pro Mini, is on pin 2.  However, on the Pro Micro, interrupt 0 is connected to pin 3 and pin 2 uses interrupt 1.  Therefore, to use the Pro Micro, you'll need to change the code in the DWM1000 library use interrupt 1.  To do this, I'll have to assume you know how to locate the folder and/or the .zip file that contains the code for this library.  Inside, you find a folder named "src" and in this folder you see a file named DW1000Ranging.cpp.

Open this file with a text editor that can edit code files, or plain text files and search for a line that reads:

DW1000.begin(0, myRST);

Edit this line to change the lone '0' to a "1" and save the file.  Close the library folder and use the Arduino IDE to recompile the sketch which should, in turn, cause the library code to be recompiled.  The modified library code should then run with the Pro Micro.  However, you'll need to undo this change if you ever want to use it with a Pro Mini.
----->

also have a look
https://github.com/thotro/arduino-dw1000/issues/158

closed 2017-04-24 by Rotzbua