thotro/arduino-dw1000 #152: Triangulation

AliSiddiqui4, 2017-01-14:
Using and the DW1000 Anchor and Tag code, I am only getting ranges from 3 anchor even though 4 anchors are connected. 

Furthermore does anyone know how to send these ranges data to MATLAB?

jmta, 2017-01-29:
You might be better off using a different programming language to work with these values as a quick google makes it seem like you can't have a serial input on matlab. I know C# does. 

You'll also be looking into doing triLATeration as opposed to triangulation, as these give you a distance not an angle, you can also trilaterate with 3 anchors.

berkerbatur, 2017-01-29:
@AliSiddiqui4 you could try following if Java works for you:
https://github.com/lemmingapex/trilateration

AliSiddiqui4, 2017-01-30:
I will try it but for now I seem to having the problem of getting Range readings on the tag from all three anchors. It gives me the range reading from 2 but very rarely gives the reading from the 3rd one.

berkerbatur, 2017-01-30:
You could exceed the reset period if you use DW1000Ranging_ANCHOR/TAG example. if your anchor count is fixed, modifying RangingAnchor/Tag examples and iterating through each anchor one by one could give you better sampling rates, you don't need to do Blink in this. 
But you have to add destination id in to your data (or another solution) to make RangingAnchor/Tag sketches work with multiple anchors.

AliSiddiqui4, 2017-01-31:
@berkerbatur can you please elaborate on this, I am very new to this and I have no idea whats in the code I just uploaded it on the arduino pro mini and ran it. Can you please explain step by step what I need to do?

christoshadjiaslanis, 2017-02-18:
Has anyone succeeded in 2D Trilateration using this library? How can you do TDoA Trilateration without a SYNC pin (since I am assuming you are using the DWM1000 modules which have a GPIO7 pin instead). Furthermore if someone has gotten the Trilateration to work, what is the measurement error for LOS conditions and what is the effective range? (I am considering researching this area for my Masters using this library).

Rotzbua, 2017-04-26:
Please move the information about TDoA to the wiki. Thanks.
duplicates #47 #77 #108 #128 #130 #152

wangwangguoguo, 2017-05-02:
"Using the DW1000 Anchor and Tag code, I am only getting ranges from 3 anchor even though 4 anchors are connected."

How can you realize that as I can only get the range between 1 anchor and 1 tag.

wangwangguoguo, 2017-05-18:
@AliSiddiqui4 Have you successed in get 3 ranges and in the TriLateration. If so, could you share your programm of the TriLateration?

werty37, 2017-12-26:
I have been able to get trilateration working. I used ESP to pass the data to a localization engine which i wrote using processing, albeit still in its early stages. I am getting decent trilateration results. I am using lemmingapex java library for the same.