thotro/arduino-dw1000 #154: decawave DW1000 device driver source code on CORTEX M3

sarwadenj, 2017-01-17:
Any body having this source code i want to try it cortex m3

data from manual--->
Decawave also provides DW1000 device driver software as source code. This source code includes a
set of API functions to initialise, configure and control the DW1000. It provides API functions for
transmission and reception, and for driving the functionalities of the IC. The DW1000 driver source code is targeted for the ARM cortex M3 but is readily portable to other microprocessor systems.

Thanks.

Rotzbua, 2017-01-17:
This library is not the library mentioned in the manual. The official driver is available on their homepage.

sarwadenj, 2017-01-18:
Hi @Rotzbua , 
please mention the link i'm not getting it.

Thanks.

sarwadenj, 2017-01-18:
https://github.com/lab11/dw1000-driver    have a look

Rotzbua, 2017-01-18:
@sarwadenj ?? simply decawave homepage: http://decawave.com/support/software

Rotzbua, 2017-04-24:
@sarwadenj Hi, can I close this issue?

closed 2017-05-19 by sarwadenj