thotro/arduino-dw1000 #155: distance precision

grex93, 2017-01-17:
HI, 
I tried both the range codes (dwm1000Rangin_anchor and the rangingAncor) to one tag and one Anchor.

I noticed that the code did from decawave is more precise than about 30 cm higher than the other. but both they have errors that are around 1 meter.
in particular I noticed that both codes with the most accurate measurements to get the mode: longDataFastLowPower, shortDataFastLowPower.
with the decawave code I get an error 60/70 cm under two meters, but enlarging the error also becomes 1 meter. with these two methods also can not get signals at distances greater than 5 meters.

It states that the antennas are not located inside boxes and are not in contact with the battery, but are positioned with the correct free space. (The distances calculated with antennas that link or not between them are the same)

Is there a way to calibrate the antennas or improve the indoor distance (even without obstacles or only with a wall) to have a maximum error of 30 cm ?? I tried to read the manual but I did not help ..

jmta, 2017-01-29:
I'm currently working on implementing a system to find the best value for each individual antenna delay, but in the mean time you should be able to match the accuracy seen from other code by changing the value seen in this line of code around 1000, in DW1000.CPP
`writeValueToBytes(antennaDelayBytes, 16384, LEN_STAMP);`
to match the number seen in the other code, you would likely be able to improve upon that too by iterating that number until it matches the value your actually measuring. Hopefully in the future there will be something you can run to give you the value you want to put in there.

Slamoth, 2017-02-16:
define a static float variable in DW1000.h and put that into a function in cpp. So you can set them in ino.

Rotzbua, 2017-04-26:
duplicate of #6

closed 2017-04-26 by Rotzbua