thotro/arduino-dw1000 #158: Has someone been successful using waynes tinkering page`s part II PCB and components?

Mraeclo, 2017-01-23:
Hey, how is it going?

I have recently tried to make a indoor localization system, based on this description 
part 1
https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000
part 2
https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

Has anyone tried this and been successful?

I got everything according to the links on part 2, from the pcb to all components and the transceiver, at digikey, except from item 1: Dual Schmitt Trigger - Buffer, wich at the table had the reference 296-13012-2-ND at digikey, but using this one wasn't possible to get less than 3k units, so I bought the 296-13012-1-ND, wich seems to be the same.

After receiving, I've paid to get them soldered at a specialized store (I have not enough soldering experience to try this at home =/ ).

Using arduinos pro mini 5v 16hz atMega 328p, and this library, I have tried to use the DW1000 anchor and dw1000 tag examples but it doesnt print anything, and when I use the Basic Conectivity Test, it prints
DW1000 initialized ...
Committed configuration ...

after that, sometimes it prints

Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

and others

Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

from reading other threads about this, when it prints the second one its supposed to be working, right? But when I turn 2 on, one using the anchor and other the tag examples, nothing happens, maybe because of this situation about the prints, not really sure, any thoughts?

any help appreciated

regards

maxicastle, 2017-01-23:
I had exactly the same problem,

First I thought it was a physical hardware problem, but i checked three different boards with the same problem, So Physical or soldered discarded. 

at this point I really do not know what was the problem, but I solved doing this

I used this version of arduino 1.6.9
I used an early version of the Arduino-Decawave-Library DW1000 Version=0.6
I modified the next File DW1000.cpp (Modify INPUT to INPUT_PULLUP)
   -Line 105	pinMode(_rst, INPUT);  to pinMode(_rst, INPUT_PULLUP);
  - Line 211 	pinMode(_rst, INPUT); to pinMode(_rst, INPUT_PULLUP);

and this works for me.

Can I ask you, Who store soldered your components? and it was expensive?

Rotzbua, 2017-01-24:
> Line 211 pinMode(_rst, INPUT); to pinMode(_rst, INPUT_PULLUP);

I do not recommend this. As discussed earlier in some issue reports the data sheet warns you about using a pullup. It maybe damage the dw because it is used as an input and an output.

See: dw1000 data sheet v2.08 §5.6.1 page 20, the RSTn pin should not be driven high but left floating.

maxicastle, 2017-01-24:
Yes, I know that recommendation, I know it's not recommended, but it works for me.

Mraeclo, 2017-01-24:
@maxicastle Thanks for the answer. I had it soldered at a store called Mestia (mestria.com.br). Its in Belo Horizonte, Brazil, so Im not sure if this can help you somehow =P. It costed about 15 dollars a piece.
Im downloading the old version and making the changes, will post result.

@Rotzbua thanks for the answer as well, I`ll give it a try, see if it works for 1 piece then the others =)

Mraeclo, 2017-01-24:
@maxicastle the line pinMode(_rst, INPUT); only exists in the newer version of the library, its pinMode(_rst, OUTPUT); at the 0.6 version.

maxicastle, 2017-01-24:
Ok I used this version only modify the extension from .png to .zip xD

[image]

Mraeclo, 2017-01-24:
@maxicastle  thanks a lot! looks like its working although only with 3.3v arduino pro mini. About the print:
Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

is it ok for the Unique ID to be like that?

when it printed DECA before using your code, the model and version was different, it was model 1 version 3, not sure if this makes any difference.

Anyways, im getting more arduinos 3.3v tomorrow and post the results here.

thanks again!

reycreator17, 2017-01-29:
Hi guys. I'm really new to this so please bear with me. I just wanna ask why are there so many components in Wayne's tinkering page. Cause I've seen this page 

http://khjtony.github.io/project/2015/05/15/Test-DWM1000-UWB-Module-with-Arduino.html

It seems like he just plug the chip directly into his arduino pins and seems like it works fine. To anyone provide me explanation would be very helpful Thanks!

jmta, 2017-01-29:
Waynes does a complete logic level shift from 3.3v to 5v and vice versa to enable communication between the 2, the page you linked might be using a 3.3v arduino as opposed to the 5v wayne used.

reycreator17, 2017-01-29:
Hi @jmta Thanks for answering.

Does it mean that I won't be able to communicate a tag which is connected to Arduino 3.3V logic level to an anchor of Arduino 5V logic level? 

sorry this is quite a dumb question so don't be harsh

jmta, 2017-01-29:
All arduinos have to be 3.3v. 5V won't be able to communicate and simply won't see anything. From a simplified technical perspective ( at least how I see it, basic knowledge of digital electronics) for the arduino to recieved a bit of 1 it is waiting to see 5v on the data pin, which never happens, therefore no data is received as everything is 3.3v

reycreator17, 2017-01-29:
Cool. May I ask you 1 more question please @jmta 

I've been planning to use DWM1000 for the navigation of my remote controlled robot. Suppose I have 3 anchors stationary and 1 tag at my RC robot. Is it possible to save the distances or let's say coordinates into a SD card?

jmta, 2017-01-29:
Its certainly possible, its something I am currently working on but not doing it onboard the arduino but using a PC connected to each using bluetooth. This link `http://math.stackexchange.com/questions/884807/find-x-location-using-3-known-x-y-location-using-trilateration` should help you. I'm working with drones so am interested in the 3rd dimension hence am subbing value back in to get a z co-ordinate (Assuming it hasn't fallen through the floor). But in your case with a robot on the floor you should be well covered by the link given.

reycreator17, 2017-01-29:
Your the best @jmta . Thanks. 
I hope they soon fix the issues of waynes adapter board so I can start writing my code and test

jmta, 2017-01-29:
@reycreator17 What arduino board are you using, I might have something that will help you out. We're currently using a 3.3.v arduino pro mini.

reycreator17, 2017-01-29:
@jmta 

I need to use Arduino Mega since I will be using lot of sensors and actuators. Sounds like I really need to learn that level shift 3.3V to 5V

Mraeclo, 2017-01-30:
@maxicastle 
Did not work using your code, but with the 1.6.9 arduino IDE and latest library, I`ve changed the 2 existing lines related to the _rts to INPUT_PULLUP and its working. thanks a lot!

pipsitel, 2017-02-11:
did you see the boards from localino.net? I have some and they work really great.

chipaudette, 2017-02-17:
I also used the latest library and simply changed all of the pinMode(_rts,INPUT) lines to be pinMode(_rts,INPUT_PULLUP).  I am now getting the Device ID response to saw "DECA".

I'm using an Arduino Pro Mini 5V with the "Wayne Holder" decawave pro mini adapter described on ttps://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii

christoshadjiaslanis, 2017-02-18:
@jmta 
For the sake of completeness, the 3.3V/5V problem is not because data cannot be read on the Arduino side, actually the 5V arduino will read 3.3V logic levels (https://www.arduino.cc/en/Reference/Constants). Higher than 3.0V will be read as logic one. The level shifting is done so that the DW1000 which operates at 3.3V is not damaged.

davepmo, 2017-03-04:
The problem is Wanyne's board drives the Reset pin with an LVC chip.  I was able to make this board work without modifying the library but cutting the trace on the bottom of the board that runs from the lvc chip to the DW1000 reset pin.  Then add a wire from the arduino pin9 directly to reset.  The arduino will then drive reset low appropriately, and when it switches to an input, it "releases" the reset pin allowing it to pull itself high.   Because the arduino never drives the pin high, it can be a 3V or 5V arduino, doesn't matter.

closed 2017-04-24 by Rotzbua

werty37, 2017-09-11:
Hi,

I was able to mod the Wayne board as suggested by [davepmo](https://github.com/davepmo) and it seems to work. Please check the update in Wiki. 

https://github.com/thotro/arduino-dw1000/wiki/Testbed-and-Adapter-board#wayne-board

Thanks
S

Xin2017, 2018-02-13:
Hi,
I have tried to use Wayne's board. I have followed all the instructions from the video, but when I connect the Arduino to the IDE, I cannot program it because the IDE cannot identify the Port. I have tried several boards, but not anyone works. Has anyone met same problems? Is my soldering problems?
Thanks.
Xin

zongbaoyang, 2018-03-27:
I also have this problem. Why?
I ordered the Wayne Board and used the "BasicConnectivityTest" example to test the printout:
DW1000 initialized ...
Committed configuration ...
Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 64 symbols (code #4), Channel: #1
SPIMISO pin default high in SPI, is it correct?

Xin2017, 2018-03-28:
I have the exactly same problems, do you have any idea?

Regards,

Xin 

From: zongbaoyang
Sent: March 27, 2018 1:59 AM
To: thotro/arduino-dw1000
Cc: Xin2017; Comment
Subject: Re: [thotro/arduino-dw1000] Has someone been successful using waynestinkering page`s part II PCB and components? (#158)

I also have this problem. Why?
I ordered the Wayne Board and used the "BasicConnectivityTest" example to test the printout:
DW1000 initialized ...
Committed configuration ...
Device ID: FFFF - model: 255, version: 15, revision: 15
Unique ID: FF:FF:FF:FF:FF:FF:FF:FF
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 64 symbols (code #4), Channel: #1
SPIMISO pin default high in SPI, is it correct?

zongbaoyang, 2018-03-29:
I have solved it and I have implemented ranging test. My problem is that the 74LVC2G17DBVRG4 pin is not soldered firmly. It is recommended that you check if the pin connections are normal according to the PCB diagram.

Xin2017, 2018-03-29:
Hi, Yang,
Thanks for your suggestions, I will check my soldering. But my problem is I can not figure out which part has the problem, I am not good at soldering.
By the way, how is your ranging test accuracy ?

Regards,

Xin 

From: zongbaoyang
Sent: March 29, 2018 3:14 AM
To: thotro/arduino-dw1000
Cc: Xin2017; Comment
Subject: Re: [thotro/arduino-dw1000] Has someone been successful using waynestinkering page`s part II PCB and components? (#158)

I have solved it and I have implemented ranging test. My problem is that the 74LVC2G17DBVRG4 pin is not soldered firmly. It is recommended that you check if the pin connections are normal according to the PCB diagram.

zongbaoyang, 2018-04-01:
Ranging accuracy within 10cm. You can check whether the 74LVC2G17DBVRG4 pin is firmly soldered to the PCB diagram by using the multimeter's resistance file and check whether the adjacent pins of the DWM1000 are shorted. It is recommended to check if the pin connections are normal according to the PCB diagram.

SATHYALAKSHMI123456, 2019-06-19:
hi
 my dwm1000 board working within 2meter.how can i do this for long distance ?