thotro/arduino-dw1000 #159: FS_PLLTUNE tunning value

gabrielortiz616, 2017-01-25:
Hi,

In DW1000.cpp tune() function (line 458) you set:

} else if(_channel == CHANNEL_5 || _channel == CHANNEL_7) {
	writeValueToBytes(fspllcfg, 0x0800041DL, LEN_FS_PLLCFG);
	writeValueToBytes(fsplltune, **0xA6**, LEN_FS_PLLTUNE);
}

According to the manual the value of FS_PLLTUNE should be set to 0xBE for channel 5 and 7. Is there a reason why it is set to 0xA6? Or it is just a typo.

Thank you!

thotro, 2017-01-25:
Hi,

oh, you're quite right there; that's a typo! :-)
You want to file a PR or let me know if I should fix it in your name?

Many thanks!

gabrielortiz616, 2017-01-25:
You can fix it.

You've done an awesome work, thank you!

Rotzbua, 2017-01-27:
@gabrielortiz616 Please reopen it, it should only be closed if it is fixed. thanks.

thotro, 2017-01-28:
Sorry for the delay and many thanks!

closed 2017-01-28 by thotro