thotro/arduino-dw1000 #160: Can I do it without adapter board?

hu265, 2017-01-28:
Hi everyone, I'm a beginner in arduino so pardon me if this sounds like a dumb question.
Currently, I want to do a simple two way ranging using "DW1000Raning_ANCHOR" and "DW1000Ranging_TAG" from this library to roughly measure two objects distance. I am currently using 2 arduino pro mini 3.3v and 2 of the Decawave DWM1000 modules. But it seems like I might need to get adapter boards since the DWM modules don't seem to be physically compatible with the arduino and I have no idea how to make the adapter boards on my own.
I've looking at Wayne's Tinkering page 
[https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii](url)
But I just wonder is it possible to connect my arduino mini to the DWM1000 without using the PCB?

jmta, 2017-01-29:
If you have a very steady hand and can solder wires directly to the chip you can do it without the board but honestly your best bet is to find a local hackspace who might be able to help you out (will also probably be a cool place to hang out) or find a company online to do it for you.

hu265, 2017-01-29:
@jmta That sounds reasonable!
It seems like you're quite familiar with the wayne's adapter board from your answer here [https://github.com/thotro/arduino-dw1000/issues/158](url) so I want to ask a few questions on that particular adapter board as well.
I'm quite confused about the type of 3.3v/5v arduinos to be used. I'm currently using 2 of the 3.3v arduino pro mini and i bought the adapter board from this page : 
[https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000---part-ii](url)
Since wayne's board has level shift from 3.3v to 5v. Does it mean that I can use 5v arduino mini without the adapter board but I should use 3.3v if I work with the board? 
Thank you!

reycreator17, 2017-01-29:
@hu265 

How do you know that the Arduino pro mini at Wayne's video is 3.3V logic level?

jmta, 2017-01-29:
@hu265  I'm not too familiar with the board TBH but understand the basic principles at least, If your using 3.3v arduinos you don't need to have any logic level shifts. This should help you understand.
`All arduinos have to be 3.3v. 5V won't be able to communicate and simply won't see anything. From a simplified technical perspective ( at least how I see it, basic knowledge of digital electronics) for the arduino to recieved a bit of 1 it is waiting to see 5v on the data pin, which never happens, therefore no data is received as everything is 3.3v`
I can provide you with some files without a logic level shifter if that would interest you. They power from a 1s quadcopter battery (going through a V reg) as the arduino doesn't provide enough power, and we also didn't want to be running loads of wires all over the room to provide power (we are going to communicate back using Bluetooth) 

@reycreator17
The arduino ISN'T 3.3v logic level. The DWM1000 IS, the arduino isn't hence the use of a logic level shifter the logic level shifter converts the 3.3v out of the DWM1000 to 5V which the arduino 'sees' then another converts the 5V to 3.3V for the DWM1000 to 'see'. Wayne talks about using a 5V arduino here `https://sites.google.com/site/wayneholder/uwb-ranging-with-the-decawave-dwm1000`

reycreator17, 2017-01-29:
Ooooh I see it. 

"The author of the open source project based his design on an Arduino Pro Mini  running at 8MHz and 3.3 volts.  The DWM1000 runs at 3.3 volts, so this avoids the need for level shifting to connect to the 5 volt logic on 16 MHz Arduino Pro Mini.  I happened to have quite a few 5 volt Pro Minis in my parts bin so I decided to go with level shifting.  However, I first built a simple breakout board for the DWM1000 so I could experiment with it using a solderless  breadboard.  Then, once I was confident my design approach would work, I designed another board to fit beneath it and allow the breakout board to plug in on top.  Here's a photo showing the completed stack of PCBs:"

@jmta

Is it okay that my 3 anchors are Arduino pro mini 3.3V and my tag is Arduino mega 5V (which of course, I will be shifting the DWM1000 to 5V). Will that work?

jmta, 2017-01-29:
@reycreator17 It 'shouldn't' do but with all these thing theres always a chance a magical gremlin can come out of the corner just to ruin your day. Certainly give it a go and see, if it works perfect if not, you just have to pick up a 3.3v arduino which hopefully shouldn't break the bank.

reycreator17, 2017-01-29:
I see. Thanks thanks a lot @jmta

hu265, 2017-01-29:
@jmta 
Since wayne's board has a level shifter, I assume that means it won't work for my 3.3v arduino? 
Do you happen to know if there's an easier way to connect my arduino with my uwb1000 besides soldering them directly with wires? (I considered wayne's board because it looks quiet easy to put the arduino and uwb1000 together by simply staking them.)

jmta, 2017-01-29:
@hu265 if your comfortable leaving your email I can forward you the plans
for the pcb I'm currently using which I designed myself. No logic shifter,
the only extra things you'll need is a vreg and a 330 ohm surface mount
resistor. It's not the easiest to solder but I've done it all by hand, just
don't have 3 cups of coffee before trying to solder ( I speak from
experience)

hu265, 2017-01-29:
@jmta That'd be awesome! Thank you so much for answering all my questions. My email is shirley.hu.com@gmail.com

reycreator17, 2017-01-30:
@jmta 

can you give me also the design please
my email is reycreator17@yahoo.com

kuek, 2017-02-03:
were you guys successfull?

closed 2017-03-22 by hu265

cocowidmer, 2017-08-28:
hey!, first of all thank you for all the work done here! and the help you have given me with this!
Sadly I'm in the same situation as hu265, I think I'll try to solder it but I don't know how the circuit/wiring would be, because in the git there's only this type of circuit (fritzing):

[image]

if you could send me the design of the PCB?? or maybe a fritzing pic (of the wiring/circuit) that would work, to solder all the cables into an arduino mini?
my email is ceciliawidmer1993@gmail.com,

I would really really apreciate it!!!,

thanks!

yogeshwaran774, 2018-01-24:
@hu265 i am also working in the project of dw1000 ranging and i am having my own pcb design with wireless charging circuit. my code is working perfectly with pro mini and decawave adapter with 3.3V supply. But if solder decawave and atmega328p into the pcb it is not sending and receiving any signal. so there may be some impedance mismatching or error in floor planning which leads to the distortion in the RF signal track to transmit or receive. and then my pcb is 2 layer with decawave is placed on the top and quite parallel axis to the back side of the board atmega328p was placed. is it make any sense to inference in the signal? so can you provide me the design to complete my project? 
email yogesh@hacklab.in

miguelrs94, 2018-03-02:
Hello @hu265 and @jmta . I was wondering if you were succesfull without the adapter board. I'm having trouble getting someone to make the adapter in a machine so I would like to try it on my own now. Would you be so kind to send me your pcb plans please? Thank you so much.  miguelrs4339@gmail.com

AdnWa, 2018-04-17:
Hi everyone,
Can anyone (who is successfully using Decawave ) tell me what is the accuracy of decawave UWB sensor ?  
and at what distance you are getting that accuracy ?
I need to implement it on a 25m x 40m area and get an accuracy of 10cm.

Thnx !

nagendra092918, 2019-05-24:
@jmta hai, can please share the pcd design of the dwm100 interfacing with arduino my mail id is 
mail: nagendra@aerologiks.com
                        thank you

repadovrha, 2019-12-03:
Hi @hu265 and @jmta . I hope you still have dwm1000 adapter part. If you do could you please send it on my email? My mail is blaz.recek@student.um.si
Thank you

AlvaroAmosH, 2019-12-05:
Hello, @hu265 @repadovrha . Could you please send the DWM1000 adapter part to me too if you still have it? My mail is alvaroamos1@gmail.com
Thank you

SrR1ck, 2021-12-09:
> @ hu265 se você se sentir confortável em deixar seu e-mail, posso encaminhar os planos para o pcb que estou usando atualmente e que eu mesmo desenvolvi. Sem shifter lógico, as únicas coisas extras de que você precisa é um vreg e um resistor de montagem em superfície de 330 ohms. Não é o mais fácil de soldar, mas fiz tudo à mão, só não tome 3 xícaras de café antes de tentar soldar (falo por experiência própria)
> […](#)

@jmta, could you please send me the adapter card? my email: hic.bebs@gmail.com