thotro/arduino-dw1000 #161: ResetInactive

yigibihe, 2017-01-30:
Hello,
`uint32_t resetPeriod = 250;`
I was wondering why reset period was kept this short? I tried with higher values and had no problems. 

Thanks

Rotzbua, 2017-01-30:
There are some open points in your issue report: File? Line of code? Reset period of what?

yigibihe, 2017-01-31:
Reset period of the DVM 1000, also named as "inactivity period" on some files. I don't think line of code relates as this a general question about the chip.

Thank you

skadiexe, 2018-02-14:
Got many of these.
To maximize polling rate i keep the reset period low at 75ms to reduce time the system waits end of the reset_period. Try 500 and 1000 ms and nothing changed.

Why this happened?

Both tag and anchor have this issue:

[image]