thotro/arduino-dw1000 #163: without connecting IRQ pin to gpio2 I'm getting all data..unaccepted

sarwadenj, 2017-02-03:
Hi,
I can't believe this without connecting IRQ pin I'm getting following data.... 

Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

sarwadenj, 2017-02-03:
Hi, 
i did some tests with dwm1000 below are some result....

with all pin connected i'm getting---->
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

2. when i removes IRQ pin then i'm getting same result.
3.  after removing IRQ pin and RESET pin i'm getting --->
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: FFFF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

PAN and Short address changed.

I can't believe after removing IRQ, RST and VCC 3.3V I'm getting 
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: 05
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: FFFF, Short Address: 05
DeöDW1000 initialized ...
Committed configuration ...
Device ID: BC95 - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 00, Short Address: FF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: BC95 - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 00, Short Address: FF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FE:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 00, Short Address: FF
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5
DW1000 initialized .

I pressed reset button...you can see Device Id become BC95 and suddenly DECA.
If i remove any SPI pin then device id becomes 00
Don't know what happening, i tried with another decawave set up too. 
I'm programming arduino pro mini using Arduino UNO.

sarwadenj, 2017-02-03:
Hi @Rotzbua , 
I uploaded BasicSender code in Pro Mini without connecting IRQ and RST pin I'm getting following on serial monitor....

### DW1000-arduino-sender-test ###
DW1000 initialized ...
Committed configuration ...
Device ID: DECA - model: 1, version: 3, revision: 0
Unique ID: FF:FF:FF:FF:00:00:00:00
Network ID & Device Address: PAN: 0A, Short Address: 05
Device mode: Data rate: 110 kb/s, PRF: 16 MHz, Preamble: 2048 symbols (code #4), Channel: #5
Transmitting packet ... #0
ARDUINO delay sent [ms] ... 29
Processed packet ... #0
Sent timestamp ... 4408860.00
DW1000 delta send time [ms] ... 4408.86
Transmitting packet ... #1
ARDUINO delay sent [ms] ... 28
Processed packet ... #1
Sent timestamp ... 4510590.50
DW1000 delta send time [ms] ... 101.73
Transmitting packet ... #2

hdineth, 2017-09-02:
Hello,
I'm trying to set basi connectivity test with a pro mini.But i'm getting just random characters (squares) in te serial monitor. Broad rates and all stuffs has checked.