thotro/arduino-dw1000 #164: How to write my own code

AliSiddiqui4, 2017-02-04:
Hi, 
I am having a lot of problems with the given example DW1000 TAG/ANCHOR Code. The anchor connect/disconnect randomly and when I connect 3 anchors I get readings from only 2 at a time. I think the problem is with the code. So can someone please give me step by step method of how I go about understanding the code given and eventually be able to write my own effective code. I am asking because I couldn't find a proper documentation on the library and its function and the given api doc doesn't help much.

KeyYD, 2017-04-22:
The user manual is essential for this, and the basic SPI controlling is necessary. You could find some help on Arduino Hacking Page to build up your library. If you wanna hacking your own library, it may be a big work. Fighting and Good luck to you

Rotzbua, 2017-04-24:
@AliSiddiqui4 The state of the libraries is updated in the readme. Mostly they are "prototypes" which means they work for a small scenario. To make them good you have to invest time, add comments and extend it.

closed 2017-04-24 by Rotzbua