thotro/arduino-dw1000 #167: getTempAndVbat and ADC readings

cimbalo, 2017-02-08:
DW1000 ADC doesn't work unless ADCCE bit and undocumented bit 9 of PMSC_CTRL0 are set to 1.
As result **getTempAndVbat** always return a fixed value based on factory calibration values.

I can provide a pull request to fix the issue.
What do you think of a function **enableADCClock** to set those two bits?
There are better implementation options?

thotro, 2017-02-08:
Hi there! Thanks for the pointer! 

I'm not sure if it wouldn't be better to just force this state anyway on initialization; so no further functions would be necessary to be explicitly called. Or do come any drawbacks/post-enable-consequences with it? 

Your choice of implementation! :-) A PR is more than welcome!

Cheers!

cimbalo, 2017-02-13:
I trace down the problem to the lde load code.
PR fix my issue by not overwrite the register state on initialization.

thotro, 2017-02-14:
Great, many thanks!

closed 2017-02-14 by thotro