thotro/arduino-dw1000 #169: Device ID: DECA - model: 255, version: 15, revision: 15

hu265, 2017-02-13:
Currently I'm using BasicConnectivityTest code and I got the following results:

Device ID: DECA - model: 255, version: 15, revision: 15
Unique ID: EUI: 255:255:255:255:255, OUI: 255:255:255
Network ID & Device Address: PAN: 65535, Short Address: 65535
Device mode: Data rate: 6800 kb/s, PRF: 16 MHz, Preamble: 128 symbols (code #4), Channel: #5

I'm using an Arduino Pro mini 3.3v to connect to the my DWM1000 module. I connected them in this following way:

DWM1000  - Arduino Pro Mini
Pin 24 - Ground
Pin 22 IRQ/GPIO8 - PIN_IRQ (pin 2)
Pin 20 SPICLK - SCK (pin 13)
Pin 19 SPIMISO - MISO (pin 12)
Pin 18 SPIMOSI - MOSI (pin 11)
Pin 17 SPICSn - PIN_SS (pin 13)
Pin 6 and VDD3V3 - Vcc (3v3)
Pin5 VDDAON  - Vcc (3v3)

I'm not sure why I don't get the normal results like other people did. (something that looks like: Device ID: DECA - model: 1, version: 3, revision: 0) Is there any problem with the way I connect my Arduino and DWM module?

Thank you

Rotzbua, 2017-02-13:
> Device ID: DECA - model: 255, version: 15, revision: 15

==> Device wrong or poorly connected.

hu265, 2017-02-13:
@Rotzbua Thank your for your response. I'm a beginner at this project and I have been following data sheet but I'm not entirely sure which pins are still needed to be connected just. I looked at the AdapterBoardTestBed diagram and tried out several different versions of connections shown by participants in other discussions but I have been getting the 255 results or all zeros. Could you tell me how to correctly connect all the pins in my case?

Rotzbua, 2017-02-17:
@hu265 No, sry, I do not have the time to look it up in detail. Just go do decawave homepage and read the data sheets and application papers.

pipsitel, 2017-03-07:
@hu265 i also had this problem. i now have boards from localino.net, ask the guys (kuek in github) if they can give you boards without dwm. they are very helpful and open minded. it saved a lot of error finding and just worked straight away.

closed 2017-03-21 by hu265