thotro/arduino-dw1000 #172: DWM1000 time (timestamp) to int/long or String

Slamoth, 2017-03-03:
Does anyone know how we can convert this DWM1000 time type to integer or long or string ?

Slamoth, 2017-03-07:
I can get close to the timestamp value but something do not fit (code is taken from visualizeDatas in DW1000Ranging.cpp)
see code and samples below
```
#define LEN_TIMESTAMP 5
byte byteTimestamp[LEN_TIMESTAMP];
.
.
DW1000.getReceiveTimestamp(byteTimestamp);
char string[90];
sprintf(string, "%02X:%02X:%02X:%02X:%02X", byteTimestamp[4], byteTimestamp[3], byteTimestamp[2], byteTimestamp[1], byteTimestamp[0]);
Serial.println(string);
```
Above code should write timestamp to hexadecimal but the last 2 digits are always wrong
DW1000 timestamp 258914084697 prints out 3C:48:7B:67:**7F** should be 3C:48:7B:67:**59**
DW1000 timestamp 259746808713 prints out 3C:7A:1D:C3:**AF** should be 3C:7A:1D:C3:**89**
DW1000 timestamp 292090652267 prints out 44:01:F5:A6:**92** should be 44:01:F5:A6:**6B**
DW1000 timestamp 324285291400 prints out 4B:80:E8:DB:**AF** should be 4B:80:E8:DB:**88**
DW1000 timestamp 356478246715 prints out 52:FF:C2:5F:**61** should be 52:FF:C2:5F:**3B**

Any help would be appreciated guys.

Rotzbua, 2017-03-07:
I do not understand exactly from where all your values come from.

I suggest the time correction causes your observed difference.

cimbalo, 2017-03-08:
@Slamoth Take a look at DW1000Time.getTimestamp() which return a int64_t

Slamoth, 2017-03-08:
Thanx cimbalo. I am using arduino pro mini so using gettimestamp is not really accurate for me since I need receive timestamp (and pro mini is 8 mhz to dwm 38 mhz ...). But i will give it a shot. thanx again.

closed 2017-03-17 by Slamoth

cimbalo, 2017-03-19:
@Slamoth MCU speed has nothing to do with DW1000Time.gettimestamp accuracy.
You can use DW1000.getReceiveTimestamp to retrive DW1000's internal clock state at rx event (as you alredy did) then tranform it to a int64 using at DW1000Time.gettimestamp().

```
DW1000Time rxtime;
DW1000.getReceiveTimestamp(rxtime);
int64 int64rxtime;
rxtime.gettimestamp(int64rxtime);
```

Slamoth, 2017-04-13:
Did not really work as expected :(

Rotzbua, 2017-04-13:
@Slamoth Go and read the documentation of the functions and understand what the class does:
https://github.com/thotro/arduino-dw1000/blob/master/src/DW1000Time.h
https://github.com/thotro/arduino-dw1000/blob/master/src/DW1000Time.cpp

ZOIDYnet, 2017-10-10:
I had the same problem. I solved it by addition of two zero bytes in the end of each message, when I was transmitting timestamps.

AhmedGamal111, 2021-04-17:
> I had the same problem. I solved it by addition of two zero bytes in the end of each message, when I was transmitting timestamps.

Please, was your problem read wrong timestamps? And how did you add two zero bytes?

Thanks for help

Slamoth, 2021-04-19:
@ZOIDYnet got the naswer