thotro/arduino-dw1000 #177: How to send the 3 range data?

AliSiddiqui4, 2017-03-18:
Ok, so I am finally getting 3 ranges value from three anchors on the tag. I can easily use these to do Trilateration, But my main controller should not be connected to the tag, since I want to track a robot which is moving. How can I change the Tag code so that I receive all the range values on a 4th UWB (with its own code), connected to the main controller?

xforus, 2017-03-21:
Hello, you could implement, in Tag code, BasicSender example. Then, in your controller with another DWM1000 implement BasicReceive example.
Every Range cycle, you have to send from Tag a new transmision to your controller.

I want to do something like you but with 3 tags and 3 anchor.
I have 3 bunnyies and want to know how far are them :)

AliSiddiqui4, 2017-03-21:
@xforus have you tried doing this with your 3 tags?

xforus, 2017-03-22:
Not yet, I'm still learning source code structure to make changes.
Is dificult to me to add a second Tag and don't know best method to interact more than one Tag with more than one Anchor. Timings are mixed and the result is wrong distances.
Need more testing and free time!!

AliSiddiqui4, 2017-03-22:
ahan, well please do share your code if you get the 2nd tag working, need it desperately. Thanks

xforus, 2017-03-24:
Hi, AliSiddiqui4
Is dificult for me to know which is the best method to implement it.
To implement more Tags, is necessary to write a system of time slots, similar used for anchors, but for tags, in order to not get more than 1 Tag talking at the same time.
I also will need help from someone with more experience!

AliSiddiqui4, 2017-03-30:
Hi @xforus 
Best of Luck! and let me know if you had any success.

AliSiddiqui4, 2017-04-02:
@xforus ok so I tried what you suggested and it worked to some extent. I am receiving the data on a receiver dwm100 from the tag dwm . But I am getting a lot of garbage values like 100 garbage for a single correct value. Not sure what the problem is. I am using the same baud rates and the same Mode.

xforus, 2017-04-04:
Hello @AliSiddiqui4 , what do you mean with garbage?
I supose you only send data from one device to another like Basic receiver/sender example...
Do you get this garbage in these examples too?

Archeresque, 2017-10-26:
@AliSiddiqui4 Did you ever get the code for the receiver working correctly? I'm building a system of 3 anchors and one tag and I want a receiver to take the data being outputted by the tag, I tried implementing the send and receive examples but the result is garbled.

AliSiddiqui4, 2017-10-27:
Yeah just try using different modes. Also, I do not recommend this whole system, I wasted 3 months to get it to work. In the end, I did get it to work but the tag needed to be within 8-10m of each anchor in order for it to get readings from all three anchors. This is a big disappointment to what the decawave website claim about the range, someone should sue them.

Archeresque, 2017-10-27:
@AliSiddiqui4 I'm using the system inside a room, so the range isn't a problem. I was able to get roughly double that range however. Is there anyway you can share your code for the receiver? I want to compare mine with yours.

AliSiddiqui4, 2017-10-27:
Its the same library code with just changes in the mode

Archeresque, 2017-10-27:
@AliSiddiqui4 You are referring to the LOWPOWER,ACCURACY, modes etc?

AliSiddiqui4, 2017-10-27:
YES

Archeresque, 2017-10-27:
@AliSiddiqui4 I tried all 6 modes but only LONDATA_RANGE_LOWPOWER allowed me to see my packet albeit the result was still garbled. Is this the mode you had success with?

mochromadhoni, 2018-03-15:
@AliSiddiqui4 I've tried 3 anchors and 1 tag with long data range accuracy. but I only get data from 2 anchors only, very rarely get data from the third anchor. if I get the data, the data is very different from the other 2 anchors. example: distance between tags and every anchor is 60cm. and data from the third anchor is 2 meters but sometimes negative. What about your data ? Did you get same thing ?