thotro/arduino-dw1000 #178: Can't get range

hu265, 2017-03-21:
I'm attempting to test 1 tag and 1 anchor with DW1000_Ranging_ANCHOR and DW1000_Ranging_ANCHOR to get distance values. However, I kept getting "Not found", "blinks" and "delete inactive device" messages on the anchor end and never received any distance results on either end.
The screenshot here shows one of my testings
[image]

------------------------------------------------------------------------------------------------

I also tested the BasicConnectivity and BasicSender/Receiver examples to make sure connections are good and the tag and anchor are able to communicate. They seems to be working fine. 

[image]

[image]

Could anyone tell me what part might have gone wrong that's causing it and how to fix it?

xforus, 2017-03-22:
Hello, I'm starting to learning code, but it seems that Tag is not receiving Ranging Init and Anchor is deleting Tag for inactive time. But it seems that you receive a frame from Tag (or another device) that is not listed in Anchor list. This is when you see "not found".
Did you change any define timings?
A good way to know code is to put printf in every action of send and receive in order to know where you are.

hu265, 2017-03-22:
@xforus Thanks for your comments. And no, I didn't change anything on the code. And when you said the Anchor list, does it mean it's predefined?
I'm still pretty new to this, so any suggestions would be appreciated!

ybc82, 2017-03-24:
Hi, this happened to me when I increased the number of anchors to 3 but not 1. Anyway, you can try what I suggested in issue #179 . Hopefully, it helps.

hu265, 2017-03-25:
Hi @ybc82 
I hope I followed your method correctly,
DW1000Ranging.h file
	static volatile uint8_t getNetworkDevicesNumber() { return _networkDevicesNumber; }; (line 96)
        static volatile uint8_t      _networkDevicesNumber; (line 126)
DW1000Ranging.cpp file 
        volatile uint8_t      DW1000RangingClass::_networkDevicesNumber = 0; (line 41)

Though, I tested the DW1000Ranging_Anchor/Tag code, and got the similar results as before.
May I ask you how you powered your device? I've been using the USB to power both of my Arduino UNO so I wonder if it might have something to do with lack of power supply that's causing my device to be "inactive"? Thank you!

hu265, 2017-03-27:
UPDATES:
I tried out the RangingAnchor/Tag examples since I'm able to read the code a little easily through them. I found out that @xforus was correct about my previous results. I added serial.print on every steps of the message sending between my anchor and tag, and I noticed -
1. Tag constantly sends POLL to Anchor. 
2. Anchor occasionally gets POLL from Tag and tries to send POLL_ACK back.
3. Tag never detects any POLL_ACK message from Anchor and Anchor never gets RANGE from Tag, so the cycle stops.

I switch over my Tag and Anchor devices to test the program but got the same results. So my guess is that, the Anchor can find the Tag but the Tag doesn't recognize the Anchor.
I also tried changing operation mode to MODE_SHORTDATA_FAST_ACCURACY etc.., still no luck. I'm not sure if anyone has had this exact issue but I'm hoping to get some different ideas for me to try out. 
Thank you!

ybc82, 2017-03-27:
@hu265 I did not do the 'line 96' change, and other changes you made were exactly the same with what I did. Besides, since you are having this problem when there are only 1 anchor and 1 tag, and I had this issue when there were 3 anchors and 1 tag, it might be possible that there are some other causes.

On my board, the USB 5V power is connected to an LDO with 3.3V output, which provides power to both the Arduino chip and the DWM1000 module. I don't know how you're powering the DWM1000 module, but it can be a problem if its power source cannot provide enough current.

Besides, I'm using Arduino Pro mini. I'm not sure if things would be different on UNO.

Btw, I tried hard-code to stop sending all blinking messages, which theoretically would be helpful to communication condition. But this did not help in my case anyway. And this solution is probably more complicated.

Thank you!

AliSiddiqui4, 2017-04-08:
@ybc82 what mode are you using and are you getting all three ranges consecutive like
from anchor 1 
then anchor 2
then anchor 3 so on

Maxai2, 2017-09-18:
Hi!
Tell me please how you connect dw1000 to arduino... I not receive any thing

hdineth, 2017-09-28:
@Rotzbua 
hello sir,
i have created 3 anchor modules and a tag and everything works fine. Whn i connect the anchors and tag, my tag recieves a flow of data like in the attached picture from each anchor at a time sequentially. All i want to do is to store these values in a 3by 1 matrix like as follows,
`float mat[3][1];
mat[0][0]=value from 1st anchor node;
mat[1][0]= value from 2nd node;
mat[2][0]= value from 3rd node;

`
likewise. but when i initilaize and save values to the matrix (inside the newRange function) it saves like follows,
mat[0][0]=value from 1st anchor node;
mat[1][0]= value from 1st node;
mat[2][0]= value from 1st node;

are there any way to correct this issue.?
[image]