thotro/arduino-dw1000 #179: [Bug fix] Potentially solved the problem of not getting 3 ranges when running 3 anchors and 1 tag

ybc82, 2017-03-24:
I have been running 3 anchors and 1 tag simultaneously so that I should be able to get three ranges and therefore the position of the tag. But I always had problems that some anchors 'died' from time to time (I can't get range reading of them from the tag; the serial port of the anchors output lots of 'Not found'; the serial port of the tag output lots of 'delete inactive device: ', '1 device added'), which other people have suffered from too.

I recently found that the solution should be that for the variable '_networkDevicesNumber' declaration in DW1000Ranging.h and initialization in DW1000Ranging.cpp, **add** a 'volatile' to it. My explanation for this is that otherwise this variable is not read correctly in the function 'addNetworkDevices' and will cause errors in the device list.

xforus, 2017-03-24:
Hello ybc82. What example code do you have problems?
Using DW1000Ranging_ANCHOR/TAG, with last version and with defaults, I never had problems receiving 3 ranges.
But I experienced problemes when I decrease cycles of POLLS between BLINKS. For example in: counterForBlink > 20, when I change to "1" . So 50% POLLS and BLINKS is needed for me.

I found the cause of problem but not solution.
Problem is because last ANCHOR to receive POLL don't receive it. Anchor transmit RANGE INIT but don't receive POLL. So TAG is waiting for last ANCHOR but never is received, so no range is transmited to other Anchors, because Tag needs all POLL ACK's to send RANGE.

I don't know if this is the same problem to you.

ybc82, 2017-03-27:
@xforus , I'm using the same example code. I'm using Arduino Pro mini and Arduino 1.6.10. People have reported that the code works well with Arduino 1.6.9 and I don't know if that's the case.

I think the cause you described is similar to mine. And I figured out that the cause of 'this cause' was that the non-volatile variable caused problems in new device registration or deletion of inactive devices.

Thank you!

Rotzbua, 2017-03-28:
@ybc82 If it fixes the problem, please make a pull request instead of an issue. So the code can be improved.

In general the code except `DW1000.h/cpp` are a proof of concept code. I already refactored and changed a lot but there is still a lot to do if you search `TODO`. I think the code do not consider about race conditions and `volatile` variables. It would be good if you can look for such variables and fix them.

xforus, 2017-03-29:
@ybc82 I have changed the variable '_networkDevicesNumber' as you said. But this don't solve my problem.

I'm using Arduino 1.6.8.
With 1.8.1 it doesn't work nothing.

I think that problem is at receiving part of Anchor.
This is my Log where you can see 3 anchors and 1 Tag. When Tag is expected to receive a POLL ACK from 3th Anchor but don't receive Poll from Tag so it doesnt' answer with a POLL ACK. Tag don't send range to other 2 Anchors because it waits to have all POLLS ACK from all Anchors:

Log at TAG part:

BLINK_SEND
ranging init; 1 device added ! ->  short:CAE7
RANGING_INIT_RECEIVED CA:E7
POLL_SEND TO 1 DEVICES
POLL_ACK_RECEIVED CA:E7
RANGE_SEND TO 1 DEVICES
RANGE_REPORT_RECEIVED
from: CAE7       Range: 0.46 m   RX power: -68.06 dBm

BLINK_SEND
ranging init; 1 device added ! ->  short:9791
RANGING_INIT_RECEIVED 97:91
POLL_SEND TO 2 DEVICES
POLL_ACK_RECEIVED CA:E7
POLL_ACK_RECEIVED 97:91
RANGE_SEND TO 2 DEVICES
RANGE_REPORT_RECEIVED
from: CAE7       Range: 0.51 m   RX power: -67.54 dBm
RANGE_REPORT_RECEIVED
from: 9791       Range: 1.06 m   RX power: -70.45 dBm

BLINK_SEND
ranging init; 1 device added ! ->  short:BAAB
RANGING_INIT_RECEIVED BA:AB
POLL_SEND TO A 3 DEVICES
POLL_ACK_RECEIVED CA:E7
POLL_ACK_RECEIVED 97:91

**(here 3th Anchor should receive a Poll and send an ACK. but it don't. So no range from Tag and
result is : inactive time)**

**(in next cycle of blink 3th Anchor receive Poll and answres correctly):**

BLINK_SEND
!!inactive time!!
delete inactive device: BAAB
ranging init; 1 device added ! ->  short:BAAB
RANGING_INIT_RECEIVED BA:AB
POLL_SEND TO 3 DEVICES
POLL_ACK_RECEIVED CA:E7
POLL_ACK_RECEIVED 97:91
POLL_ACK_RECEIVED BA:AB
RANGE_SEND TO 3 DEVICES
RANGE_REPORT_RECEIVED
from: CAE7       Range: 0.51 m   RX power: -67.36 dBm
RANGE_REPORT_RECEIVED
from: 9791       Range: 0.99 m   RX power: -70.16 dBm
RANGE_REPORT_RECEIVED
from: BAAB       Range: 0.99 m   RX power: -68.09 dBm

kuek, 2017-03-29:
hi, I found out that it works well when compiling the code with Atmel Studio. It does not work when compiled with Arduino IDE. So it seems like a compiler problem to me.

AliSiddiqui4, 2017-04-01:
@kuek did you uploaded the original code using Atmel Studion? and are you receiving all the 3 ranges form the three tag? and is the mode the default LONGDATA_RANGE_ACCURACY?

rbrtwtrs, 2017-04-01:
I changed the variable _networkDevicesNumber to volatile and confirmed that I do now get ranges from all three anchors. I notice that the data rate varies and it appears to stumble which as Rotzbua says seems like it is a race condition. Changing to a volatile is a kludge fix to the underlying problem. 
I am using the Visual Micro Plugin for Visual Studio. If a particular compiler version does work it is also a likely kludge fix to the underlying issue.

AliSiddiqui4, 2017-04-09:
@ybc82 WHAT MODE ARE YOU USING?

ybc82, 2017-04-10:
@AliSiddiqui4 I'm not sure what you mean by 'MODE', but generally, I'm using the demo code of DW1000Ranging.

christoshadjiaslanis, 2017-04-11:
@rbrtwtrs Did you change the _networkDevicesNumber only in the TAG or also in the Anchors?

reycreator17, 2017-04-18:
Hi I managed to make 3 anchors and 1 tag run perfectly, but not so perfect with the precision when they are very near. I made my own perfboards and uses the same schematic but I power the whole thing with 3.3V parallel to the Vcc of pro mini and Vcc of DWM1000. I've used 10K ohm resistor and used some capacitors. I found out that measurements get very inaccurate when DWM1000 gets hotter, yes, when I made trials on a sunny outside. 

Also I ON first just 1 tag and 1 anchor, after they are connected, I ON another anchor and see if the 3 are connected and ON the last anchor

I want to know how to make their addresses as Anchor 1, Anchor 2 and Anchor 3 and not some just random numbers. 

Thanks Rey
[image]
[image]
[image]
[image]
[image]
[image]
[image]

kuek, 2017-04-18:
Looks like some effort, but very nice. Glad it works!

christoshadjiaslanis, 2017-04-19:
@reycreator17  nice. It seems from your pictures that you are getting negative ranges. In order to achieve static device address go to startAsAnchor in DW1000Ranging.cpp and change
`	_currentShortAddress[0] = 1;
	_currentShortAddress[1] = 1;`

What code did you use for 3 Anchors and 1 Tag?

reycreator17, 2017-04-19:
@christoshadjiaslanis

Hi. I follow your instructions and its now working now as I want. 

[image]

I'm using the code in the latest dwm-master.zip, the DW1000Ranging_ANCHOR and DW1000Ranging_TAG.

I haven't changed any in the code except in the filter

 DW1000Ranging.attachInactiveDevice(inactiveDevice);
  //Enable the filter to smooth the distance
    DW1000Ranging.useRangeFilter(true);

I enabled the filter. 

About the negative distance, yes, I'm getting negative distances when the anchors are really near to the tag. Also, it is not very accurate about 2 meters distance and I noticed that the range changes whenever I'm rotating the tag or anchor. 

I don't really know what problem you are facing, but I suggest you use arduino 1.6.8 version. I've tried 1.6.13 before and and other versions too, it doesn't really work well. In 1.6.13, I experienced, working well with 2 anchors but whenever I ON the 3rd anchor, the serial monitor just spamming "ranging init; 1 added device .... delete inactive device" again and again. Other versions, doesn't really give synchronize output in the serial monitor.

Uninstall your current Arduino, delete Arduino15 and library. Run ccleaner before installing 1.6.8. I've just follow this guy and it really worked for me 
[CLEAN UP BEFORE INSTALLING ANOTHER COPY OF THE IDE OR ARDUINO CREATE.pdf](https://github.com/thotro/arduino-dw1000/files/938480/CLEAN.UP.BEFORE.INSTALLING.ANOTHER.COPY.OF.THE.IDE.OR.ARDUINO.CREATE.pdf)

Thanks

Rey

kuek, 2017-04-20:
Your ranging inaccuracy might be caused by the disturbed RF propagation. If your antennas lie flat on a table this is happening. There has to be some air gap between the antenna and any other material.

christoshadjiaslanis, 2017-04-20:
@reycreator17  yes the ranging is a problem. I tried all day today and yesterday to fix it but I found no good solution for maintaining accuracy while supporting three anchors.

reycreator17, 2017-04-21:
@kuek 

Do you mean that there should be no board at the back of the antenna? Like what Wayne did : https://www.youtube.com/watch?v=ECUdwzs8Sc8

Should I cut my board for better accuracy?

kuek, 2017-04-21:
Or like mine... https://www.youtube.com/watch?v=Ggv8hrLanRo

air gap is important for RF propagation.

reycreator17, 2017-04-21:
@kuek 

Hi kuek. Thank you for your advise. May I know how accurate are your tag and anchors? Mine is about +- 50 cm.

reycreator17, 2017-04-21:
Do you have any idea how can I cut my adapter board without removing the DWM1000 ? I've ordered the board from OSH park and it was pretty thick.

Rotzbua, 2017-04-26:
@reycreator17 Please stay focused on the issue. We have already some topics about precision. If you have some new results from a experiment please add them to the wiki. I prefer the knowledge focused to a single point of truth :)

kuek, 2017-04-26:
Hi,
i get less than 10cm accuracy with my localinos. The results are already online.

christoshadjiaslanis, 2017-04-26:
@kuek how did you get 3 accurate ranges at the same time? I can only get one and then when I introduce more anchors I get ridiculous ranges

reycreator17, 2017-04-27:
Hi.

@christoshadjiaslanis  Are you using the adapter board here?

christoshadjiaslanis, 2017-04-27:
@reycreator17  yes I am. Also I tried using atmel studio to program instead of Arduino IDE still no luck

reycreator17, 2017-04-28:
Hi @christoshadjiaslanis, can you give me your email please. I want to ask about compiling at atmel studio. I've tried it yesterday and I'm having errors. I see that we both need for our thesis. 

Also, have you tried cutting the board at the back of the antenna?

kuek, 2017-04-28:
Hi guys, Are you sure all your anchors have unique IDs?

reycreator17, 2017-04-28:
Hi @kuek 

How to know ?

christoshadjiaslanis, 2017-04-28:
@reycreator17  it is a bit too late now. The deadline is really soon. I downloaded an extension for atmel studio called VMicro (http://www.visualmicro.com/page/Arduino-for-Atmel-Studio.aspx). Im pretty sure it uses your default arduino compiler though... (gcc) not sure. Also the board I am using is the default one in the project, it does not cover the back of the antenna

@kuek yes all the anchors have unique static ID's. Did you change the network devices number to volatile as is recommended at the beginning of this thread?

wangwangguoguo, 2017-05-15:
@reycreator17 
Could you please share me your codes who can get 3 different range shown in your picture? I've used the initial code, but il won't work.
[image]

wangwangguoguo, 2017-05-17:
@ybc82 
[image]
2 questions

1：
I've changed the code in DW1000.cpp
`volatile uint8_t      DW1000RangingClass::_networkDevicesNumber = 0;`
in DW1000.h
`static volatile uint8_t      _networkDevicesNumber;`

But as is shown in the picture the 3rd anchor can't get connected. Have I changed the code correctly?

2: 
The short address is random all the time although I define it like this:
`	_currentShortAddress[0] = 1;
	_currentShortAddress[1] = 0;`
[image]

I can't find where is wrong. What do you think?

KeyYD, 2017-05-17:
@wangwangguoguo It is better to use a stable version of Arduino IDE, such as version 1.6.6

KeyYD, 2017-05-17:
@reycreator17 Your DW1000 PCB looks nice. But the position of DW1000 may cause much more error. The User manual gives a suggestion. It will be better to make a keep-out area on you board, considering that the reason is the Antenna. 

> For best RF performance, ground copper should be flooded in all areas of the application board, except in the areas marked “Keep-Out Area”, where there should be no metal either side, above or below (e.g. do not place battery under antenna).

[image]

wangwangguoguo, 2017-05-18:
@KeyYD 
[image]
I've used 1.6.6 but it seems that nothing changed. Nothing gets wrong but the 3rd range just won't appear.

reycreator17, 2017-05-20:
Hi @KeyYD Thanks

Actually I've emailed the decawave engineers about that, if it is really needed to cut the board at the back of the antenna. I showed them the pictures of my setup. And they said it should not matter as long as there is no metal or copper on the keep out area.

reycreator17, 2017-05-20:
@wangwangguoguo 

[DW1000Ranging_TAGfinal.zip](https://github.com/thotro/arduino-dw1000/files/1015824/DW1000Ranging_TAGfinal.zip)

[DW1000Ranging_ANCHOR_final.zip](https://github.com/thotro/arduino-dw1000/files/1015825/DW1000Ranging_ANCHOR_final.zip)

sarwadenj, 2017-05-20:
Hi,
@reycreator17 , @wangwangguoguo 
Currently i made changes in "DW1000Ranging.cpp"
volatile uint8_t DW1000RangingClass::_networkDevicesNumber = 0;
in "DW1000Ranging.h"
static volatile uint8_t _networkDevicesNumber;

i have three anchors 1,2,5 all three getting added I'm getting ranges from 1 and 5 not getting from 2.
[image]
[image]

wangwangguoguo, 2017-05-21:
@reycreator17 
Thank you for your codes and I got a problem when run the DW1000Ranging_Anchor
[image]
In the library, DW1000Ranging.cpp, the function` startAsAnchor` has only two input items.
[image]
Could you also share the DW1000Ranging.cpp with me?

reycreator17, 2017-05-22:
@wangwangguoguo 

[arduino-dw1000-master.zip](https://github.com/thotro/arduino-dw1000/files/1018156/arduino-dw1000-master.zip)

reycreator17, 2017-05-22:
@sarwadenj 

Use 1.6.8

Rotzbua, 2017-05-22:
Stop sharing code. If you have fixes make a pull request so everybody using this library has a benefit. Otherwise I have to take action :/

reycreator17, 2017-05-22:
@Rotzbua

Copy. Sorry about that

Maxai2, 2017-09-15:
Hi  @reycreator17 
Can u tell me how I must connect arduino Uno and decawave dw1000? which pins?

AMUHENDIS, 2018-04-21:
Hey guys,
I could use DWM1000 as transmitter and frames is arriving correctly to Tx,Rx ..
But When Im using the distnace functions .. Status Registers is always wrong !

Tx can't receive back again from Rx !!
Any Suggestions ?
Thanks