thotro/arduino-dw1000 #180: RangingTAG.ino personal Parameter

DrummyFloyd, 2017-03-28:
Hi, 

i'm working on the DWM1000 module and i've some trouble.
when i want to set some value like 
`  DW1000.newConfiguration();
    DW1000.setDefaults();
	DW1000.setChannel(DW1000.CHANNEL_1);
	DW1000.setDataRate(DW1000.TRX_RATE_110KBPS);
	DW1000.setPulseFrequency(DW1000.TX_PULSE_FREQ_64MHZ);
	DW1000.setPreambleLength(DW1000.TX_PREAMBLE_LEN_4096);
	DW1000.setPreambleCode(DW1000.PREAMBLE_CODE_64MHZ_9);
    DW1000.setDeviceAddress(2);
    DW1000.setNetworkId(10);
    DW1000.commitConfiguration();`

it seems like my parameter are not set , because i don't have any response.

Maybe i forgot something, like some register? beacuse when i go through `DW1000.EnableMode(....)` the programm go trhought something else ?? 

thank you for your work and your time !

DrummyFloyd, 2017-04-04:
UP

Rotzbua, 2017-04-26:
This issue is confusing and makes no sense...

closed 2017-04-26 by Rotzbua