thotro/arduino-dw1000 #181: multiple Tags - best way to achieve it

prideofisland, 2017-03-30:
Hey, 
My friend and I are struggling with implementation of multiple Tags in DW1000. Actually, we have already done it once by switching the channel. It goes as follows:

- 3Anchors always use channel 5 
- Tag No.1 starts on channel 5, Tag No.2 on channel 4. 
-Tag No.1 does ranging with all 3 Anchors and switches to channel 4  
-Tag No.2 switches to channel 5 and does ranging with Anchors. If it collects measurements from all 3 Anchors, Tag No.2 switches to channel 4. 

In general, it works. The main drawback is time resolution which really sucks. System initialization after channel switching lasts about two seconds.  

We want to try sth different. We tought about dividing tags to time slots or introducing a variable 'it s your turn'.  Do you have some better ideas than that? 

Best regards,
Piotr

xforus, 2017-03-30:
Hello @prideofisland, I'm also interested in implementing more tags.

As you said, best way was dividing tags in time slots in order to have only one Tag speaking in one time.
But first you must know how many tags there are in system, and then, assign an order number to each tag to speak to anchors. There are some ways to do... It will be perfect to do a dinamically system, not close, that every tag will be able to know this information by itself...

I have found this protocol I think is very interesting for this system. It's call Token Ring Protocol:
http://www-verimag.imag.fr/~tripakis/papers/its01.pdf

prideofisland, 2017-04-03:
Hi @xforus 
> As you said, best way was dividing tags in time slots in order to have only one Tag speaking in one time.

You're right. It would be a good solution but the time resolution will be two times lower when we want to use two tags. 

> I have found this protocol I think is very interesting for this system. It's call Token Ring Protocol
Thanks for the link. 

We are going to implement something different. As you now Tag saves all Anchors in networkDevicesNumber variable. We want to add the similar feature to Anchors (they will save messages from few Tags). 

```
//each devices have a different reply delay time.
			_networkDevices[i].setReplyTime((2*i+1)*DEFAULT_REPLY_DELAY_TIME);
```
To get the system working we have to set other replay delay time for Tags too. 
[image]

xforus, 2017-04-04:
Hi @prideofisland, I'm happy to know you also want implement more tags. I would more time to dedicate with this but only have some time at some weekends.

Time ago, I also tried to do what you say about doing a list of tags in anchor device, but it don't work for me. There are more thing to keep in mind. Tags finally speaks in a kaos way... 

I think that Tags have to sincronize between them. If there is a master Tag, this is easy...
But I dont want to have a Master and what I have now is a system that every Tag generates an order number position reading its own short adress and from other tags. In this way every tag knows when start to do blinks or polls.

Tell me if you get something with method you say using reply delay time

prideofisland, 2017-04-04:
Hey @xforus . 
That's nice to hear. Hope, we ll have more luck. 
After two days of coding it can certainly be said that it's really complex thing to do. Despite this,we managed to establish first connection. 
[image]
Our system is composed of two Tags(B1 and B3) and one Anchor(A2). 

1. Anchor receives two BLINKs from B1 and B2
2. Anchor sends one Ranging_INIT to both of them (broadcast)
3. Anchor receives two Poll messages from each of Tags
4. Anchor sends one PollAck message (broadcast to all Tags)
5. Anchor receives first Range(from B1 for example) msg and calculates the distance
6. Anchor receives second Range(from B3 for example) msg and calculates the distance
Measurement starts from third point. 

There are still a lot of things to improve. I hope that I can share my results with you in the following week.

xforus, 2017-04-10:
Hi @prideofisland , this is perfect! Is exactly what I wanted to achieve.
Are you using an old MAC version of code?
Could you give me more details or make a branch? I would like to try your method and see what I could do..
Thanks!

KeyYD, 2017-04-22:
Hi@prideofisland 
Good to hear your idea. I am building one anchor and multi tag connections, too. The broadcast way is exactly what I reflect these days. I consider that this way:
the anchor broadcast to multi tag and receives there message to control which distance to measure.
It sounds like that the teacher call a student within lots to answer the question on class. (But I am not sure whether the response time is satisfying. )
:)

xforus, 2017-04-29:
Hi! I'm trying to add more than one Tag but I can't, I don't know how to control times.
I have one Tag speaking with up to 6 Anchors. But when I try to coordinate more Tags, system is going too bad...

@prideofisland , @KeyYD Do you have better results? Could you give me more details how to implement this?

Thank you!!!

prideofisland, 2017-05-02:
Hi @xforus and @KeyYD ,
Yes, we managed to achieve the 2 Tags ranging with 3 Anchors. System works like as illustrated above. We set a fixed delay time for each Tag (7ms for Tag1 and 21ms). You have to do it in order to prevent signal collisions (microprocessor needs some time to process the signal. First thing you have to do is analayse MAC frames and find there timestemps. 

2 Tags+3 Anchors --> 6 measurements in one cycle.  Our data rate is about 20 measurements pro s. So about 3,3 cycles/s. It's not much faster than 'token' strategy but much more complicated to implement. It seems to me that it was not a perfect decision. 
We're going to upload a short video with our sensors. I'll post a link here.

prideofisland, 2017-05-03:
Here is the video:
[youtube](https://www.youtube.com/watch?v=_37_zuBgMl0&feature=youtu.be)

KeyYD, 2017-05-04:
Happy to hear @prideofisland .
@prideofisland @xforus My system contains two tag and one anchor. And it can work, too! I didn't use POLL_ACK message but broadcasting way with time line controlled. 
Some of the point should be cared:
Software:
1. it will be better that the addresses of tag or anchor is decided by you, and not randomly.
2. some part of library code should be adjusted to run the multi tag/anchor.
3. the time line should be controlled.
Hardware:
1. the current while working in the receiving mode is about 0.12A (You can see this in the picture)
Here is my test version of anchor. (the DC regulated power supply shows that the current is about 120mA)
[image]
Still hacking
: )

kuek, 2017-06-11:
we also implemted multi-tag. this will be available on the website in the next days.
https://youtu.be/vubk64ok1TA

prideofisland, 2017-09-23:
> Could you give me more details or make a branch? I would like to try your method and see what I could do..

Hey,
Some people ask me to upload me code. I'm ready to do it. Should I make a branch? or a new repository? There are pretty much changes in code, new unit etc. I have also some MATLAB code for user interface and reducing errors. I've finished my adventure with DW1000 chip and will probably never again work with it.  Anyway, I want to share my knowledge (and code) to help you develop this amazing project.

Rotzbua, 2017-09-24:
@prideofisland Sounds good, the best way would be to split your new stuff into different topics and according branches.

For example I suggest following topics:
- the **driver** for the dw are the files `DW1000.h/cpp` and `DW1000Time.h/cpp`
- the **package** to build correct frames/packages `DW1000Mac.h`
- the build in **ranging** of the class `DW1000Ranging.h` and `DW1000Device.h`
- alternative ranging as example `DW1000Ranging_**`
- examples for the library

hdineth, 2017-09-28:
hello sir,
i have created 3 anchor modules and a tag and everything works fine. Whn i connect the anchors and tag, my tag recieves a flow of data like in the attached picture from each anchor at a time sequentially. All i want to do is to store these values in a 3by 1 matrix like as follows,
`float mat[3][1];
mat[0][0]=value from 1st anchor node;
mat[1][0]= value from 2nd node;
mat[2][0]= value from 3rd node;

`
likewise. but when i initilaize and save values to the matrix (inside the newRange function) it saves like follows,
mat[0][0]=value from 1st anchor node;
mat[1][0]= value from 1st node;
mat[2][0]= value from 1st node;

are there any way to correct this issue.?
[image]

AlexisTM, 2017-11-17:
To use multiple tags localization on the same band I made a "Synchronizer". One tag that triggers the other tags in a synchronous way.

kerr4huo, 2017-11-18:
@AlexisTM , I am also exploring on similar ideas now but hasn't found a solution. If possible, could you share some references you have used, please?
Thanks.

AlexisTM, 2017-11-18:
I am using Pozyx devices (DWM1000 + IMU).

I am not sure it is part of the default Firmware; but the Pozyx allows to call functions remotely. 

Therefore I have one more device I am just using to synchronise. I send to the devices: Take your position measurement. The the foreign robot, I have a Python script polling for the interruption bit. When there is an interruption, I read the position. 

Yet, the GPS like solution is the one which is scalable. Instead of ranging, you can send position and timestamps. Then, you have 2 solutions:

* All devices are well synchronised in time: you have a direct measurement of time with the timestamps received and with 3 anchors position/timestamp you have your position. 

* Only your anchors are synchronised in time: you need to solve the time ambiguity thus need at least 4 anchors (just like we need 4 satellites to get a position)

robintzeng, 2018-01-16:
Hello ~@prideofisland  
I see that  you did some amazing project
I want to do something like you recently   
,but when i download your project  (multiple_Tags)
I cannot use it 
the two tags can not work at the same time !!
Can you please teach me how to use it ???

Note: i add the missing DW1000.cpp by myself because i see you deleted it 
 and use the example DW1000Ranging_TAG

AlexisTM, 2018-01-16:
@robintzeng The way I'm doing it for N tags is having N+1 tags.

N tags are waiting and the "+1" is triggering them one by one at a certain frequency. Therefore, only one tag at the same time is using the channel.

robintzeng, 2018-01-18:
hello !! @AlexisTM  thanks for your reply 
I tried to change the channel to prevent the signal from being interrupted 
but when  I used 2 Anchors on channel 5, 1 tag  on channel 5, 1 tag on channel 3 
they couldn't communicate well.
And when I turned off the channel 3 tag
the tags and Anchors on channel  5 can work well .
Do you have any idea to deal with it  ??
and can you tell me what channel  you set the N+1 tag and your tags and Anchors on ??  
Thanks !!

AlexisTM, 2018-01-18:
I.never said anything about channels, keep the same channel for all!

robintzeng, 2018-01-18:
[image]

@AlexisTM  
Is the system I drawn same as the system you talk about ???

AlexisTM, 2018-01-18:
Except that it is all on the same band.

AlexisTM, 2018-01-18:
Yes; with 4+ anchors for 3D and 3+ anchors for 2D

pipsitel, 2018-01-18:
Like it. I will also try this also with my additional Localinos I have soon.
The channel thingy makes sense for sure ;-)

Aaron101010, 2018-01-20:
[image]

Right now I am testing the library for multiTags with all the anchors and tags on same channel.  My understanding of the UWB is that 
1. On one channel only one tag can polI all the anchors at the same time, then the anchor pollAck
2. Then tag add the anchor's Id into the network, and recognize anchors. And anchors also add tag into the network.
3. Then tag transmit poll to all the anchors again.
4. Anchor transmit Range message to the specify tag ID .  
5. Tag calculate the TOF and print the range value.
So I just modified the code like the picture, trying to use Serial input to control the poll from tag. In my test, once I Serial input '0' to stop communication, then both the anchor and tag delete Device Id, Then I want to restore the poll, I Serial input '1', but can't restore.  How can I control the poll from tag? 
By the way, I use Channel_5.
I think once I can control the poll from tag then I can design a Synchronizer with Timer to control multiTags. 
All suggestions are welcome! 
Thanks!

And I still confuse about different channel's communication.  Is channel's bandwidth overlapping, then can communicate with each other? or the central frequency?  or preamble codes? 
Really needs help!

[image]

AlexisTM, 2018-03-26:
@robintzeng Reverse TDOA implementation: The anchors are messaging and the tags are passive (only listen). This means you can have **as many tags as you want**.  

This is done by the Loco positioning system by Bitcraze for CrazyFlies, using the DWM1000.  

Anchor software:
- *anchor.c*: https://github.com/bitcraze/lps-node-firmware/blob/master/src/uwb_tdoa_anchor2.c

Tag software:
- *tag.c*: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/src/lpsTdoaTag.c
- *tag.h*: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/interface/lpsTdoaTag.h
- *config*: https://github.com/bitcraze/crazyflie-firmware/blob/a4dc79f53af771788c8fc67275b07b29ccc7a3a1/src/deck/drivers/src/locodeck.c

Arsenezhu, 2018-04-27:
I put three Anchors around the 2m*3m competition venue。 
(0,0)
(2,0)
(1,3)
with height of 0.35m
And the tag is on the robot with height 0.35m。The picture below is one of the 3 Anchors
[image]

The test video is with only 3 anchors and 1 tag in channel_7.
https://drive.google.com/open?id=1TxSGrx_w0xiMKlBBKkp-kTj5YrbTJhDZ 
It works great.

Then I am trying to implement 6 Anchors 2 tags system. 3 Anchors and 1 tag are all in channel_7 and the other set of 3 Anchors and 1 tag are all in channel_4.  Both tags works well with around 3 minutes. Then one set of the tag starts not to receive one of the 3 Anchors in same channel.
Below picture is the two Anchors. I put 2 Anchors in same point.
(0,0) (0,0)
(2,0) (2,0)
(1,3) (1,3)
with height of 0.35m and 0.3m
[image]

I think maybe the problem is I put two anchors too close? And the electromagnetic wave is too strong, so it interrupt?

Arsenezhu, 2018-04-27:
Hellow  @prideofisland @kuek @AlexisTM. Above is the problem I encounter.
Any advice is welcome. Really thanks!!!!

AlexisTM, 2018-04-28:
I've no clue on perturbation. Yet, the antennas are not omnidirectional. For best results, use the antennas straight. (Up to 5dBm difference) 

Also, you can just do as I said before: start the localisation from a master device.

Arsenezhu, 2018-04-28:
@AlexisTM  Thanks for your advice. I know what you suggest. But I think that your device is all on the same channel with the view to minimize the number of DWM module used. What about the update rate with your multi tags?
And even I try one tag and one anchor on channel_7  and another  tag and Anchor on channel_4
they still seems to interrupt after about ten minutes later. Have you ever meet this kind of problem?
Thanks for your advice.

AlexisTM, 2018-04-28:
The rate is divided by N tags. 

The other solution is using the Bitcraze firmware and have as many tags as you want.

TimotF, 2018-05-06:
@Arsenezhu I used 4 anchors and 2 tags for locating my robots within the 2*3m area. I used the same channel for every device and implemented tag synchronization as follow :

- when one tag is turned on, it waits for another tag to blink. If after some time, nothing is received, this tag becomes the master tag. 
- The master tag computes the ranging and send a sync message for each slave tag one after another.

This is a kind of multiple tag connection with a dynamic master tag. the update rate i get is about 6 to 9 Hz, which is enough for locating robots considering their max velocity.

xingshanxi, 2018-06-03:
Hi everyone!
I am puzzled with channel and I do not know how to select specific I want.
is anybody know that.
thanks.

xingshanxi, 2018-06-03:
Hi everyone!
I am puzzled with channel and I do not know how to select specific channel I want.
is anybody know that.
thanks.

carbonadam, 2019-02-07:
@ AlexisTM Did you attempt to get the bitcraze code working at all?  Im so curious to see their system in action.

AlexisTM, 2019-02-07:
@carbonadam They recently released the "roadrunner", a crazyflie with the DWM1000 without motors: https://store.bitcraze.io/collections/positioning/products/roadrunner

I did not try it yet but we are planning on buying one setup for AGVs.

Two videos they published:

https://www.youtube.com/watch?v=eWKTFT0CnU4

https://www.youtube.com/watch?v=hIH3bwq2B6U

carbonadam, 2019-02-08:
Yes there whole system looks great.  I know that they used the code here to make their system so it should be possible to recreate the tdoa 2 or 3 ourselves no ?  Decawave also has the mdek1001 for 270 euros and I thought to try it just to get some stable test results to start. Apart from a good crystal oscillator and sync Im not sure why doing the hardware ourselves is so so hard