thotro/arduino-dw1000 #182: DW1000RangingClass::startAsTag includes an analog read to an unspecified pin

crnorris, 2017-03-30:
Hi,

The function:

`void DW1000RangingClass::startAsTag(char address[], const byte mode[])`

includes the line:

randomSeed(analogRead(0));

This line was causing the rest of my sketch to not work properly by driving A0 low, which I'm using for something else.

Rotzbua, 2017-03-30:
@crnorris What do you purpose with this issue? This seems to be a specific problem of your code and not of the library...

crnorris, 2017-03-30:
Well the trouble is the library is hard-coded there to use pin 0 for the random seed. The other pins the libary uses are not hard-coded, but this one is, and it took a little bit of investigation to find out what was going on.

An easy fix would be to just add the pin number to the parameters of that function, at least that way anyone using the function knows the function is going to do something with that pin, so they can choose an unused one.

stuartaw, 2017-03-30:
The point is probably that the library doesnt make it clear that this 
pin is 'consumed' like this..

It is probably not the best way, really.. it is a very non-obvious side 
effect, which is never desirable.

Rotzbua, 2017-04-26:
duplicate of #79
Dont complain, fix it, make a pull request!

closed 2017-04-26 by Rotzbua